prusa3d/PrusaSlicer #12256: Error when starting prusaslicer

Wikingus, 2024-02-15:
### Description of the bug

when running PrusaSlicer newer 2.52 .error:This program requires UTF-8 locale to run.
and the program does not start.
while version 2.52 works ok.
Arch Linux operating system, Polish language

### Project file & How to reproduce

[Zrzut ekranu z 2024-02-15 16-15-55.png.zip](https://github.com/prusa3d/PrusaSlicer/files/14298620/Zrzut.ekranu.z.2024-02-15.16-15-55.png.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.1

### Operating system

archlinux

### Printer model

slicer on my computer

Leone25, 2024-02-26:
I found a solution, it's just a workaround but i guess it works.

Basically the issue is caused by prusa slicer expecting `.UTF8` at the end of the `$LANG` env variable.
Knowing this the solution is quite easy:
1. If you are starting prusa slicer from the terminal you can run `export LANG="$LANG.UTF8"`, then you can run `prusa-slicer` and it will run.
2. If you use manjaro kde, in the applications menu, you can right click on the prusa slicer link and click `edit application`, then under the `application` tab enter `LANG=en_US.UTF-8`(you can change the language here, unfortunately the variable trick doesn't work) in the environment variables and click ok. Now starting prusa slicer from the applications menu won't give the error, and it will run properly.

hope this helps someone

Wikingus, 2024-03-07:
It worked, I had an error in the locale.gen file and it was generating incorrectly. now it is working. Thanks 
file locale-gen :
```
en_US.UTF-8 UTF-8
pl_PL.UTF-8 UTF-8
pl_PL ISO-8859-2
```
```
[wiki@Archlinux prusa3d_linux_2_7_2]$ locale
LANG=pl_PL.UTF-8
LC_CTYPE="pl_PL.UTF-8"
LC_NUMERIC="pl_PL.UTF-8"
LC_TIME="pl_PL.UTF-8"
LC_COLLATE="pl_PL.UTF-8"
LC_MONETARY="pl_PL.UTF-8"
LC_MESSAGES="pl_PL.UTF-8"
LC_PAPER="pl_PL.UTF-8"
LC_NAME="pl_PL.UTF-8"
LC_ADDRESS="pl_PL.UTF-8"
LC_TELEPHONE="pl_PL.UTF-8"
LC_MEASUREMENT="pl_PL.UTF-8"
LC_IDENTIFICATION="pl_PL.UTF-8"
LC_ALL=pl_PL.UTF-8

```

closed 2026-09-29 by prusaslicer-bot[bot]