prusa3d/PrusaSlicer #12257: No Z lift when ramping lift disabled - collision danger!!!

vovodroid, 2024-02-15:
### Description of the bug

When ramping lift disabled nozzle makes direct movement to new point with some final elevation:

[image]

Last one object point:

[image]

First second object point:

[image]

### Project file & How to reproduce

[Shape-Cylinder.zip](https://github.com/prusa3d/PrusaSlicer/files/14299045/Shape-Cylinder.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.2-alpha2+win64

### Operating system

Win10x64

### Printer model

Biqu B1 SE

rtyr, 2024-02-15:
> When ramping lift disabled

Ramping lift is enabled in your project file (via filament overrides). I don´t see any issue.

vovodroid, 2024-02-15:
I reupload the projects with overrides disabled, and lift height 1mm. 

[image]

[image]

The same single movement from 8mm to 9mm:

[image]

SachCZ, 2024-02-16:
Hello @vovodroid, even thought there was initially some confusion with filament overrides, **I was able to reproduce your issue**. It is indeed a bug. And we are already working on resolving the issue. Thank you for using alpha version of Slic3r and helping us catch bugs such as this one in the early stages of public testing.

lukasmatena, 2024-02-20:
Fixed in [2.7.2-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-beta1). Closing.