prusa3d/PrusaSlicer #12258: [BUG] No slowdown when Z travel speed is zero.

vovodroid, 2024-02-15:
### Description of the bug

When Z travel speed is zero no cooling layer slowdown is applied.

[image]

Z travel is not zero:
[image]

[image]

Z travel is zero:
[image]

[image]

 

### Project file & How to reproduce

[Shape-Cylinder.zip](https://github.com/prusa3d/PrusaSlicer/files/14299438/Shape-Cylinder.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.2-alpha2+win64

### Operating system

Win10x64

### Printer model

Biqu B1 SE

rtyr, 2024-02-15:
Reproduced, thanks.

lukasmatena, 2024-02-16:
SPE-2163

lukasmatena, 2024-02-20:
Fixed in [2.7.2-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-beta1). Closing.

vovodroid, 2024-02-20:
@lukasmatena , wouldn't it be mo reasonable to use `machine_max_feedrate_z` instead of `travel_speed`? Anyway it will be limited by max. Z-speed.

lukasmatena, 2024-02-20:
@vovodroid Maybe it would. The hack with zero speed was introduced three years ago for backwards compatibility, when I separated the travel z speed from the normal speed (bd6badf and 4657e0d). With the default set to zero, old profiles in the new slicer should do exactly the same as before. You may be correct that the case which I wanted to be compatible with probably does not occur in practice.

rtyr, 2024-02-20:
machine_max_feedrate_z is available only with some g-code flavors