prusa3d/PrusaSlicer #12260: Add option to disable top surface infill tracing around a raised object (i.e. logo)

ddanderson1, 2024-02-15:
When adding a logo to a flat solid object, prusaslicer will trace around the base of the logo on the final top surface layer under the raised logo.  The issue is that this creates surface imperfections in the top surface since the infill can't keep a constant pattern under the logo (starts and stops in pattern).  Even with Monotonic lines top fill pattern selected, the "lawn mower" effect is still visible.

There is a current workaround to partially fix this by changing to 100% infill and changing the top layers to 0.  Height range modifiers can be added to reduce the infill percentage to less than 100% and ironing can still be enabled on the topmost surface.  This now allows the top surface under the raised logo to be printed in one single pattern with no breaks and no tracing around the logo.  However the issue is the print speed for the top layer matches the infill speed (200 mm/s) and there's no way to override this.  So the surface is improved, but due to the print speed at 200 mm/s, it creates an undesirable matte finish.

dougemon, 2024-02-21:
Hi there,
I experience the same issue and I wanted to add some context. I am attaching a 3mf project that describes the scenario above.
[Raised Text Infill.zip](https://github.com/prusa3d/PrusaSlicer/files/14354110/Raised.Text.Infill.zip)

[image]
[image]
[image]

To answer your question about the speed, I found that you could actually change the Height Range Modifier value under Speed>Infill to the desired slow speed.

But what we really need in order for the height range modifiers to work is the ability to select Monotonic Rectilinear as an internal infill pattern.
Hope this makes sense, I would prefer a toggleable option here for those top surfaces to decide whether to "look ahead" or not.

k1-801, 2024-03-04:
I think this issue is partially related to #9533 (and the mentioned issue could benefit from this fix)

Tupson444, 2024-03-04:
> NOT break the infill line if the resulting gap is less than a certain distance

This is a good idea

ddanderson1, 2024-04-27:
Agree with the other suggestions here for slicer improvements to deal with this.  Another workaround I have found is to increase the number of perimeters until the logo doesn't have any infill.  Only really practical if the logo is really thin, otherwise the added perimeters is generally undesired.

mvirkkunen, 2024-12-17:
@k1-801 Hey, I think I came up with the same idea as you the other day and made a PR for it. If you know how to compile from source, would you mind giving #13775 a try and seeing if it works for you?

k1-801, 2026-09-18:
> @k1-801 Hey, I think I came up with the same idea as you the other day and made a PR for it. If you know how to compile from source, would you mind giving #13775 a try and seeing if it works for you?

My apologies, somehow I didn't get a notification. I moved on to Orca, but I'll try your PR and compare it to latest mainline when I have the time (probably next week)