prusa3d/PrusaSlicer #12263: [Feature request] single wall first layer ☑️

zuidwijk, 2024-02-15:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

No, there’s no problem

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I’d like to see an option in the PrusaSlicer to choose for a single wall/perimeter on the first layer (and top surface would be handy too). 

Now you have to use shape modifiers, and though it works, it would be way easier if you could just select it as option. 

**Describe how it would work**
Be as descriptive as your skill level allows you.

Maybe the same way as OrcaSlicer, somewhere in the settings as a checkbox. 

[image]

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Current alternatives is using shape modifiers, but that more time consuming. A simple checkbox ☑️ would be so much easier. 

**Additional context**
Add any other context or paint drawing about the feature request here.

neophyl, 2024-02-16:
Duplicate issue.  See https://github.com/prusa3d/PrusaSlicer/issues/11654 which is itself a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7986 etc

This has been requested multiple times now.  I believe it is on the roadmap from this comment https://github.com/prusa3d/PrusaSlicer/issues/11965#issuecomment-1872160218 in yet another closed duplicate of this feature request.

zuidwijk, 2024-02-16:
My bad for the duplicate... Seeing the other issues from 2023 and 2022,... I hope it'll be implemented soon.

Jan-Soustruznik, 2024-02-23:
Closed as duplicate https://github.com/prusa3d/PrusaSlicer/issues/7986

closed 2024-02-23 by Jan-Soustruznik