prusa3d/PrusaSlicer #12268: Opening a project with empty platter from file explorer does not work

ghylander, 2024-02-16:
### Description of the bug

If I save a project with empty platter, PrusaSlicer detects this and warns me.

I intended to use these empty projects as templates or shortcuts to specific configurations without bloating the configurations library.

However, these empty platter projects do not open from the file explorer. They only open if I start a PrusaSlicer instance and then open it from that instance.

### Project file & How to reproduce

[Template.zip](https://github.com/prusa3d/PrusaSlicer/files/14308155/Template.zip)

Simply try to open it from explorer.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1+win64

### Operating system

Windows 10

### Printer model

ender3 s1 pro

ak1n, 2025-04-01:
While from within a PrusaSlicer instance I was previously able to open empty projects (I also saved configs in 3mf files because it's way easier than managing from within prusa slicer), now attempting to open my previous config 3mf files I get an error "The supplied file couldn't be read because it's empty" which has completely borked my organized workflow.

neophyl, 2025-04-02:
The workaround until it gets fixed is to add a small object, mark it as non printing and then stick it somewhere out the way.  

The bonus feature for doing this is you can use the object as a library to 'store' commonly used modifier settings.  You can then copy/paste from the small 'library' object to your actual objects.

Jan-Soustruznik, 2025-04-03:
Hello,Thank you for your report. We will try to fix this in one of the upcoming release. An internal issue has been created (SPE-2736).

TerPuc, 2025-04-04:
Fixed in [2.9.2-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.2-rc1). Closing.

closed 2025-04-04 by TerPuc