prusa3d/PrusaSlicer #12269: Impossible print because of too few organic supports

ErwanAliasr1, 2024-02-16:
### Description of the bug

I wanted to print a small piece from https://www.printables.com/model/168815-miniature-amstrad-pc1640-ntp-deskclock, put the piece at a given angle to get a smooth surface, and ask for organic supports.

Prusaslicer 2.7.1 and 2.7.2-alpha1 offered me an impossible configuration to print by having too few organic supports.

I expected more organic support to make this part printable. 

### Project file & How to reproduce

[impossible_print.zip](https://github.com/prusa3d/PrusaSlicer/files/14311766/impossible_print.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2-alpha1

### Operating system

Linux on Mac Mini M1

### Printer model

MK4

Area5142, 2024-02-16:
To address the support for the section at that specific angle, consider adjusting the Overhang threshold angle to 45, deviating from the default 40:
[image]

By making this adjustment, you should obtain support for the entire model:
[image]

Alternatively, you can utilize the 'paint on support' feature to manually place support where needed.

In my experience, automatic support generally works well for most models, but it may encounter challenges with unique cases. Experimenting with different support types (grid, snug) or making slight adjustments to support parameters often resolves such issues.

starthal, 2024-03-22:
Possible duplicate of #11841

closed 2026-09-29 by prusaslicer-bot[bot]