prusa3d/PrusaSlicer #12271: STP import/slicing error

smurfix, 2024-02-16:
### Description of the bug

Some models are printed with filled holes on the first layer.

### Project file & How to reproduce

Take the file Nextruder/idler-lever-b-r2.stp from the MK4 upgrade. Observe two holes.

[image]

Slice it. Observe one filled hole.

[image]

Project file:

[proj.zip](https://github.com/prusa3d/PrusaSlicer/files/14313980/proj.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1 a2, 2.6

### Operating system

Linux Debian

### Printer model

MK3

neophyl, 2024-02-16:
The model has errors, it has the yellow warning triangle next to it in the list.  Also on your first image of the model, see the blue edges, those are problem edges.  Which tallies with the reported 271 open edges.
[image]

Even the built in windows fix tool cant repair the model, it fills it in just like when you slice.  The inside of the hole at both the top and bottom aren't connected to anything and are just open.  

When fixed the model slices correctly
[image]

[idler-lever-b-r2_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/14316487/idler-lever-b-r2_fixed.zip)

smurfix, 2024-02-16:
*Sigh* thanks. Then maybe somebody should fix the files in https://www.printables.com/model/451501-mk4-printable-parts/ — I have to admit that I did not expect to see broken STEP files for your own printer. :-P

closed 2024-02-16 by smurfix

neophyl, 2024-02-16:
Not my printer.  I don't work for Prusa.

smurfix, 2024-02-17:
Sorry @neophyl I didn't mean you specifically when I wrote "you" but the folks at Prusa who may or may not read this.

lukasmatena, 2024-09-12:
Fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases).

adelton, 2025-01-02:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/12122: what is the canonical reproducer? I see the `idler-lever-b-r2_fixed.stl` from https://github.com/prusa3d/PrusaSlicer/issues/12271#issuecomment-1949407007 sliced with two holes even with a build with Fedora's stock opencascade 7.8.1-3.fc41.x86_64 build.

And the `idler-lever-b-r2.3mf` from https://github.com/prusa3d/PrusaSlicer/issues/12271#issue-2139036118 has that orange exclamation mark triangle and only one through hole is in the sliced output even with a build which uses OCCT 7.6.1.

What is a canonical repoducer?

adelton, 2025-01-02:
I do see a difference with the`Spray Can 218x153 5.step` from https://github.com/prusa3d/PrusaSlicer/issues/13138#issuecomment-2243405787 where with a built using OCCT 7.6.1has five cones and no orange exclamation mark triangle is shown, while with opencascade 7.8.1 it only has four and the orange exclamation mark triangle is shown. 

Is that the reproducer and confirmation I'm after?