prusa3d/PrusaSlicer #12272: ERROR:  Variable Layer Height is not supported with Organic supports  even when NO supports are included

segallis, 2024-02-16:
### Description of the bug

I get this error on ALL models with either Variable Layer Height tool or Height Range Modifier even when there are NO supports at all.  Running PrusaSlicer 2.7.1 on Linux.   "Generated Support Material" unchecked.  "Auto Generated supports" unchecked.  All options for support material greyed out.  Support Material Extrusion width is greyed out.

### Project file & How to reproduce

Any model will give this error, with all supports disabled.
[VarLayHeightBug.zip](https://github.com/prusa3d/PrusaSlicer/files/14314134/VarLayHeightBug.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.1+linux-x64-GTK3-202312121425.AppImage

### Operating system

Ubuntu 22.04.3

### Printer model

EnderV3 KE

Jan-Soustruznik, 2024-02-23:
Hello, @segallis , 

I checked your 3MF file. You have enabled "Enforce support for the first" x Layes that means that the supports are enabled. Please check the attached link with more info about the feature: https://help.prusa3d.com/article/support-material_1698#enforce-support-for-the-first

Jan-Soustruznik, 2024-02-23:
Closed as solved

closed 2024-02-23 by Jan-Soustruznik