prusa3d/PrusaSlicer #12275: Color change for dual head XL doesn't select proper head

radzimir, 2024-02-16:
### Description of the bug

For Prusa XL I set color change for second extruder for some layer. 

Slicer produces code like:

```
;COLOR_CHANGE,T1,#00FF00
M601
M117 Change filament for Extruder 1
M486 S0
```

It doesn't work as expected from user perspective.
At the begining of the layer, printer makes simple pause and if extruder 1 is still active, 
going to menu "Settings->Change Filament" changes for still active extruder 1, not 2 as selected in slicer.

Color change pause needs to be executed only after tool change within a given layer was started.

Additional, separate problem, is that color change for multiple extruder system is apearently not labeled in any way on the vertical layers-bar, so one need to move a mouse over all layers and look on the mouse-over-popups to find one.

### Project file & How to reproduce

[ColorCangeXL_SecondHead.zip](https://github.com/prusa3d/PrusaSlicer/files/14316926/ColorCangeXL_SecondHead.zip)

[image]

See layer at 1mm height, there is color change for extruder 2 set.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.1+win64

### Operating system

Windows 10

### Printer model

Prusa XL dual head, firmware 5.1.2

rtyr, 2024-02-17:
This has been adressed in PS2.7.2-alpha2. See https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-alpha2

radzimir, 2024-02-17:
Ok, is duplicate of #11517
I tested it now with PS2.7.2-alpha2 and it work perfectrly, thanks.

closed 2024-02-17 by radzimir