prusa3d/PrusaSlicer #12281: [BUG] Infill doesn't touch perimeters

vovodroid, 2024-02-18:
### Description of the bug

With certain model/config infill if printed in the middle of the inside space:

[image]

### Project file & How to reproduce

[191.zip](https://github.com/prusa3d/PrusaSlicer/files/14321568/191.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.2-alpha2+win64

### Operating system

Win10x64

### Printer model

Biqu B1 SE

Jan-Soustruznik, 2024-02-20:
Hello, @vovodroid , 

What is your expected result? According to your setting, the infill is generated inside Modifier only. So infill is generated according to your setting.

vovodroid, 2024-02-20:
Something like this:

[image]

[image]

Modifier includes the whole object width, so I would expect infill touching perimeter.

beng27, 2024-02-20:
I think not a bug. The infill is only allowed to occupy the modifier. So it behaves as though there was a solid wall at the edge of the modifier. To be like the pink lines you have drawn, it would have to change angle. Not possible. To see what I mean, if you increase infill to a higher %, or a different pattern, you will get more of the effect you want.
In this photo I have changed your infill to 40%; the results are what I would expect.
[image]

Jan-Soustruznik, 2024-02-21:
But you have also set the Fill density = 9% So slicer calculated that the infill will fill 9% only of selected area. So that is reason why the infill doesnt touch the perimeter. Extend the fill density or use another infill pattern. 

I closed as solved. infill is generated according to user settings. 
> Something like this: [...]