prusa3d/PrusaSlicer #12286: Please add a way to paint a streight line

cozmo14047, 2024-02-19:
Hi there, at the momment when you use the brush option you have to free hand it and it can look messy. Please add an option to draw a streight line. Eg holding shift will draw a streight line. As you can see below free handing looks messy.
[image]

neophyl, 2024-02-19:
This is a duplicate issue.  Actually its a duplicate multiple times.

https://github.com/prusa3d/PrusaSlicer/issues/9765
https://github.com/prusa3d/PrusaSlicer/issues/7059
https://github.com/prusa3d/PrusaSlicer/issues/6557
https://github.com/prusa3d/PrusaSlicer/issues/5970

closed 2024-02-19 by rtyr

lukasmatena, 2024-02-19:
@neophyl Thanks for the summary. I closed all of them except for #5970.

cozmo14047, 2024-02-19:
Ok, its just that this was for multicolor painting nkt seam painting

lukasmatena, 2024-02-19:
@cozmo14047 Yes, that does not make a difference. The painting tool is the same, when it gets implemented, it will be implemented for all of them.