prusa3d/PrusaSlicer #12288: Wipe on inner perimeters instead of outer perimeters

fidoriel, 2024-02-19:
**Is your feature request related to a problem? Please describe.**
Depending on the filament quality, `wipe while retracting` (the wiping move) can have a visible impact to the most outer shell.
[image]

**Describe the solution you'd like**
I think this could be solved by not wiping into the outer shell but into the inner shell, like the red line on the image:
[image]

Also wiping into infill would be great.

**Describe how it would work**
Add a new checkbox `Do not wipe on outer shells` or `Wipe into inner peremiters`. Checking this will do the wiping move and the zhop after that on a inner perimeter or infill if possible.

Jan-Soustruznik, 2024-02-20:
Hello, @fidoriel , 

could you share your 3MF file, please? Thank you

fidoriel, 2024-02-20:
[wipe example.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14348068/wipe.example.3mf.zip)

closed 2026-09-29 by prusaslicer-bot[bot]