prusa3d/PrusaSlicer #12291: [Request] Windows 11 ARM64 build

eblieb, 2024-02-19:
Requesting a Windows 11 Arm64 build. VS should be able to compile for Windows 11 Arm64 without much issue.

JustinBeckerMSFT, 2024-02-21:
I second this suggestion. The only limiting factor I see is the kernel drivers for updating printer firmware would need to be ported.

It also looks like PrusaSlicer might have some prebuilt x64 dependencies, which isn't a problem: you could use our ARM64EC technology for that https://learn.microsoft.com/en-us/windows/arm/arm64ec

hmartinez82, 2024-03-03:
@eblieb @JustinBeckerMSFT I'm going to give it a try. I have a Windows Dev Kit and my personal laptopt is a Surface with an ARM64 cpu

hmartinez82, 2024-03-04:
Some news.
While building the dependencies, only three packages failed. Glew, Boost and OpenEXR:

Glew fails to link with:
```
LINK: command "C:\PROGRA~1\MICROS~1\2022\PROFES~1\VC\Tools\MSVC\1439~1.335\bin\HOSTAR~1\arm64\link.exe /nologo CMakeFiles\glew.dir\C_\Dev\Github\PrusaSlicer\deps\build\dep_GLEW-prefix\src\dep_GLEW\src\glew.c.obj CMakeFiles\glew.dir\C_\Dev\Github\PrusaSlicer\deps\build\dep_GLEW-prefix\src\dep_GLEW\build\glew.rc.res /out:bin\glew32.dll /implib:lib\glew32.lib /pdb:bin\glew32.pdb /dll /version:2.2 /machine:ARM64 /INCREMENTAL:NO -BASE:0x62AA0000 -nodefaultlib -noentry libvcruntime.lib msvcrt.lib opengl32.lib glu32.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST:EMBED,ID=2" failed (exit code 1355) with the following output:
   Creating library lib\glew32.lib and object lib\glew32.exp
LINK : fatal error LNK1355: invalid base address 0x62AA0000; ARM64 image cannot have base address below 4GB
ninja: build stopped: subcommand failed.
```

Boost fails while building Boost Context and doing some arm assembling. ARMASM does not accept -D defines like MASM does:
```
[1/264] Building ASM_ARMASM object libs\context\CMakeFiles\boost_context.dir\src\asm\ontop_arm64_aapcs_pe_armasm.asm.obj
FAILED: libs/context/CMakeFiles/boost_context.dir/src/asm/ontop_arm64_aapcs_pe_armasm.asm.obj
C:\PROGRA~1\MICROS~1\2022\PROFES~1\VC\Tools\MSVC\1439~1.335\bin\HOSTAR~1\arm64\armasm64.exe -DBOOST_CONTEXT_EXPORT="" -DBOOST_CONTEXT_NO_LIB -DBOOST_CONTEXT_SOURCE -DBOOST_CONTEXT_STATIC_LINK -IC:\Dev\Github\PrusaSlicer\deps\build\dep_Boost-prefix\src\dep_Boost\libs\context\include -IC:\Dev\Github\PrusaSlicer\deps\build\dep_Boost-prefix\src\dep_Boost\libs\assert\include -IC:\Dev\Github\PrusaSlicer\deps\build\dep_Boost-prefix\src\dep_Boost\libs\config\include -IC:\Dev\Github\PrusaSlicer\deps\build\dep_Boost-prefix\src\dep_Boost\libs\core\include -IC:\Dev\Github\PrusaSlicer\deps\build\dep_Boost-prefix\src\dep_Boost\libs\static_assert\include -IC:\Dev\Github\PrusaSlicer\deps\build\dep_Boost-prefix\src\dep_Boost\libs\throw_exception\include -IC:\Dev\Github\PrusaSlicer\deps\build\dep_Boost-prefix\src\dep_Boost\libs\mp11\include -IC:\Dev\Github\PrusaSlicer\deps\build\dep_Boost-prefix\src\dep_Boost\libs\pool\include -IC:\Dev\Github\PrusaSlicer\deps\build\dep_Boost-prefix\src\dep_Boost\libs\integer\include -IC:\Dev\Github\PrusaSlicer\deps\build\dep_Boost-prefix\src\dep_Boost\libs\type_traits\include -IC:\Dev\Github\PrusaSlicer\deps\build\dep_Boost-prefix\src\dep_Boost\libs\winapi\include -IC:\Dev\Github\PrusaSlicer\deps\build\dep_Boost-prefix\src\dep_Boost\libs\predef\include -IC:\Dev\Github\PrusaSlicer\deps\build\dep_Boost-prefix\src\dep_Boost\libs\smart_ptr\include -IC:\Dev\Github\PrusaSlicer\deps\build\dep_Boost-prefix\src\dep_Boost\libs\move\include  -o libs\context\CMakeFiles\boost_context.dir\src\asm\ontop_arm64_aapcs_pe_armasm.asm.obj C:\Dev\Github\PrusaSlicer\deps\build\dep_Boost-prefix\src\dep_Boost\libs\context\src\asm\ontop_arm64_aapcs_pe_armasm.asm
Microsoft (R) ARM Macro Assembler Version 14.39.33521.0 for 64 bits
Copyright (C) Microsoft Corporation.  All rights reserved.

error A2029: unknown command-line argument or argument value -DBOOST_CONTEXT_EXPORT=

 Usage:      armasm [<options>] sourcefile objectfile
             armasm [<options>] -o objectfile sourcefile
             armasm -h              for help
```

OpenEXR fails when trying to use intrinsics. 
```
[3/33] Performing build step for 'dep_OpenEXR'
FAILED: dep_OpenEXR-prefix/src/dep_OpenEXR-stamp/dep_OpenEXR-build C:/Dev/Github/PrusaSlicer/deps/build/dep_OpenEXR-prefix/src/dep_OpenEXR-stamp/dep_OpenEXR-build
C:\Windows\system32\cmd.exe /C "cd /D C:\Dev\Github\PrusaSlicer\deps\build\builds\OpenEXR && "C:\Program Files\CMake\bin\cmake.exe" --build . --config Release -- -j6 && "C:\Program Files\CMake\bin\cmake.exe" -E touch C:/Dev/Github/PrusaSlicer/deps/build/dep_OpenEXR-prefix/src/dep_OpenEXR-stamp/dep_OpenEXR-build"
[1/83] Building CXX object OpenEXR\IlmImf\CMakeFiles\IlmImf.dir\ImfVersion.cpp.obj
[2/83] Building CXX object OpenEXR\IlmImf\CMakeFiles\IlmImf.dir\ImfPreviewImage.cpp.obj
[3/83] Building CXX object OpenEXR\IlmImf\CMakeFiles\IlmImf.dir\ImfDwaCompressor.cpp.obj
FAILED: OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfDwaCompressor.cpp.obj
C:\PROGRA~1\MICROS~1\2022\PROFES~1\VC\Tools\MSVC\1439~1.335\bin\HOSTAR~1\arm64\cl.exe  /nologo /TP  -IC:\Dev\Github\PrusaSlicer\deps\build\builds\OpenEXR\OpenEXR\IlmImf -IC:\Dev\Github\PrusaSlicer\deps\build\dep_OpenEXR-prefix\src\dep_OpenEXR\OpenEXR\IlmImf -IC:\Dev\Github\PrusaSlicer\deps\build\builds\OpenEXR\OpenEXR\config -IC:\Dev\Github\PrusaSlicer\deps\build\dep_OpenEXR-prefix\src\dep_OpenEXR\IlmBase\Iex -IC:\Dev\Github\PrusaSlicer\deps\build\builds\OpenEXR\IlmBase\config -IC:\Dev\Github\PrusaSlicer\deps\build\dep_OpenEXR-prefix\src\dep_OpenEXR\IlmBase\Half -IC:\Dev\Github\PrusaSlicer\deps\build\dep_OpenEXR-prefix\src\dep_OpenEXR\IlmBase\Imath -IC:\Dev\Github\PrusaSlicer\deps\build\dep_OpenEXR-prefix\src\dep_OpenEXR\IlmBase\IexMath -IC:\Dev\Github\PrusaSlicer\deps\build\dep_OpenEXR-prefix\src\dep_OpenEXR\IlmBase\IlmThread -external:IC:\Dev\Github\PrusaSlicer\deps\build\destdir\usr\local\include -external:W0 /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MD /O2 /Ob2 /DNDEBUG /EHsc /showIncludes /FoOpenEXR\IlmImf\CMakeFiles\IlmImf.dir\ImfDwaCompressor.cpp.obj /FdOpenEXR\IlmImf\CMakeFiles\IlmImf.dir\IlmImf.pdb /FS -c C:\Dev\Github\PrusaSlice [... cut ...] ssor.cpp
C:\Dev\Github\PrusaSlicer\deps\build\dep_OpenEXR-prefix\src\dep_OpenEXR\OpenEXR\IlmImf\ImfName.h(113): warning C4996: 'strncpy': This function or variable may be unsafe. Consider using strncpy_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
C:\Program Files\Microsoft Visual Studio\2022\Professional\VC\Tools\MSVC\14.39.33519\include\emmintrin.h(20): fatal error C1189: #error:  This header is specific to X86, X64, ARM64, and ARM64EC targets
```

JustinBeckerMSFT, 2024-03-04:
- It looks like the issue with Glew is known and has been resolved in more recent revisions: https://github.com/nigels-com/glew/pull/384 I wonder if PrusaSlicer takes a hard dependency on the Glew version in the dependencies or if pulling a new version would be sufficient?

- Looking at the ASM, it looks like those defines aren't used anywhere in the file. It may be correct to just remove those -D options from the ARMASM command line.

- What target architecture are you using for this build? If OpenEXR uses Intel intrinsics, and doesn't have an ARM64 intrinsic equivalent implementation, then you will need to build as ARM64EC to get those intrinsics.

hmartinez82, 2024-03-04:
> * It looks like the issue with Glew is known and has been resolved in more recent revisions: [Base address only for 32bit nigels-com/glew#384](https://github.com/nigels-com/glew/pull/384) I wonder if PrusaSlicer takes a hard dependency on the Glew version in the dependencies list or if pulling a new version would be sufficient?

That's awesome. If just a version bump is enough that'd be perfect. I'll try that in my next round of attempts.

> * Looking at the ASM, it looks like those defines aren't used anywhere in the file. It may be correct to just remove those -D options from the ARMASM command line.

Not trivial, it's not like there's code in PrusaSlicer's CMake files to explicitly add those -D defines. :( .  I'll have to look at the Boost CMake input params to see why this is happening. Boost supposedly already support Windows on ARM for a while so I'd be suprise if there's a legit issue with Boost itself and ARMASM (crossing fingers here).

> * What target architecture are you using for this build? If OpenEXR uses Intel intrinsics, and doesn't have an ARM64 intrinsic equivalent implementation, then you will need to build as ARM64EC to get those intrinsics.

This is the most tricky. I'll take a look too, but I know we are building Glew for Windows on ARM, for a while now btw, in MSYS2 (https://packages.msys2.org/base/mingw-w64-glew so not exactly the same thing because MSYS2 uses Clang with GCC's frontend)

hmartinez82, 2024-06-18:
It would be nice if the the PrusaSlicer team updated the dependencies to versions that support ARM64 then I could give it a try.

hmartinez82, 2024-07-03:
@enricoturri1966 @bubnikv Who can I ask to update and Boost? I'm almost in the finish line with building PrusaSlicer for Windows on ARM64. 
The only offending 3rd party libraries are:
- OpenEXR
- Boost
- Glew

Newer versions of Boost already fixed the only issue I see with building it for ARM64
OpenEXR is super old. The deps folder is using version 2.5.5, but OpenEXR is already at 3.2.4.
I'll create a pull request for GLEW shortly.

JustinBeckerMSFT, 2024-07-03:
@hmartinez82 Since there are third party dependencies that aren't built for ARM64, you should look into ARM64EC (https://learn.microsoft.com/en-us/windows/arm/arm64ec) this technology will let you mix-and-match ARM64 and x64 code. This can all be done natively inside of Visual Studio. Please let me know if you have any questions :)

hmartinez82, 2024-07-17:
@JustinBeckerMSFT No can do. Everything is statically linked. At least that's what I noticed.

But I think I got the Boost patch working.

JustinBeckerMSFT, 2024-07-17:
Hi @hmartinez82 thanks for continuing to look into this. Can you elaborate on why you think static linking will not allow building as ARM64EC? One of the main purposes of developing ARM64EC was to allow static linking of mixed architecture libraries. If something in the docs is misleading, please let me know so we can try to get it fixed on our side.

hmartinez82, 2024-07-17:
> Hi @hmartinez82 thanks for continuing to look into this. Can you elaborate on why you think static linking will not allow building as ARM64EC? One of the main use cases of ARM64EC is to allow static linking of mixed architecture libraries. If something in the docs is misleading, please let me know so we can try to get it fixed on our side.

Oh. I thought that you could only use x64 code from an ARM64EC process if it the x64 code was in a DLL!! , but I have good news and bad news
- Good news. I got all the dependencies building, I have three pull requests to address them
- Bad news. MPFR and GMP are actually not built, there are binaries present in the repository and they are just copied to the output!! I can't for the life of me find a way to build their original source codes with Visual Studio, so I may, after all, look into building PrusaSlicer as ARM64EC. The PRs are: https://github.com/prusa3d/PrusaSlicer/pull/13108 https://github.com/prusa3d/PrusaSlicer/pull/13109 and https://github.com/prusa3d/PrusaSlicer/pull/13110

hmartinez82, 2024-07-17:
@JustinBeckerMSFT I guess I didn't notice this second table! 🤦🏼 
[image]

hmartinez82, 2024-07-18:
@JustinBeckerMSFT ARM64EC is a no go. Some of the dependencies are refusing to be found by PrusaSlicer. I would bet they are being built with ARM64 even when switching to ARM64EC in the top deps build list file.

It turns out that vcpkg has both mpfr and gmp so I'm going to build native ARM64 of those there!

hmartinez82, 2024-07-18:
[image]
Success!

hmartinez82, 2024-07-18:
[image]

hmartinez82, 2024-07-18:
> I am interested in what was required for the ARM64 port to work (with the intention of [porting the patches over to OrcaSlicer](https://github.com/SoftFever/OrcaSlicer/issues/5783)). Is there a list of tasks that I can refer to?

Check the pr: https://github.com/prusa3d/PrusaSlicer/pull/13116

hmartinez82, 2024-07-18:
> Thanks. One question though: would bumping up the dependency versions instead of applying patches work equally as well?

Yes for Boost . I didn't check the other ones.

Sent with Shortwave <https://www.shortwave.com?utm_medium=email&utm_content=signature&utm_source=aGVybmFuLmMubWFydGluZXpAZ21haWwuY29t>

eblieb, 2024-07-18:
> @JustinBeckerMSFT I guess I didn't notice this second table! 🤦🏼 [...]

Isn't this the whole reason for ARM64X :)
https://learn.microsoft.com/en-us/windows/arm/arm64x-pe

eblieb, 2024-07-18:
> > I am interested in what was required for the ARM64 port to work (with the intention of [porting the patches over to OrcaSlicer](https://github.com/SoftFever/OrcaSlicer/issues/5783)). Is there a list of tasks that I can refer to?
> 
> Check the pr: #13116

... This was excatly why I started this whole thread, to hopefully get to the point that OrcaSlicer can also benefit :)

JustinBeckerMSFT, 2024-07-18:
> Isn't this the whole reason for ARM64X :) https://learn.microsoft.com/en-us/windows/arm/arm64x-pe

Yes and no. ARM64X is more for building DLLs that can be loaded into both ARM64 native and x64 emulated processes. It can also be used to build executables that run on both x64 and ARM64 machines natively, but that use case is not formally documented.

Congrats to @hmartinez82 for the successful port! My one concern from here is the UX around the printer drivers. If you offer PrusaSlicer ARM64 in the installer it prompts you to install the drivers for updating printer firmware. These drivers run in kernel mode and therefore cannot be emulated. To ship this to ARM64 users I imagine the installer would either need to block installation of the drivers on ARM64 machines, or someone would need to port the drivers.

hmartinez82, 2024-07-18:
> > > @JustinBeckerMSFT I guess I didn't notice this second table! 🤦🏼 ![image](https://private-user-images.githubusercontent.com/1100440/349654488-b668b82c-3326-4476-b89d-49e6e753e621.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEzMDY4NTAsIm5iZiI6MTcyMTMwNjU1MCwicGF0aCI6Ii8xMTAwNDQwLzM0OTY1NDQ4OC1iNjY4YjgyYy0zMzI2LTQ0NzYtYjg5ZC00OWU2ZTc1M2U2MjEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MThUMTI0MjMwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9N2IwYzcyZThiNzQxYjYxNDM1ZTA1MzBjMzcxNTU1ZDM5MTFjMTk3MzMyZGI0YzZlMjNkYTgzYWFlMTBjMjE0ZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.uYCh1vOai7qXOmJv_tCP0GeXw9XpCxa4bs-_Oi7p3Wc)
> > 
> > 
> > Isn't this the whole reason for ARM64X :) https://learn.microsoft.com/en-us/windows/arm/arm64x-pe
> 
> Yes and no. ARM64X is more for building DLLs that can be loaded into both ARM64 native and x64 emulated processes. It can also be used to build executables that run on both x64 and ARM64 machines natively, but that use case is not formally documented.
> 
> Congrats to @hmartinez82 for the successful port! My one concern from here is the UX around the printer drivers. In the PrusaSlicer installer it prompts you to install the drivers for updating printer firmware. These drivers run in kernel mode and therefore cannot be emulated. To ship this to ARM64 users I imagine the installer would either need to block installation of the drivers on ARM64 machines, or someone would need to port the drivers.

For sure, the work is not done. I just wanted to get it building locally as easy as possible. The driver part is the one part I can't help with because I know nothing about driver development :(

JustinBeckerMSFT, 2024-07-18:
@hmartinez82 I took a look online and I think that the drivers are closed source so we wouldn't be able to do much about it anyways. Also, judging from the .inf files that PrusaSlicer deploys in C:\Program Files\Prusa3D\PrusaSlicer\Rambo the drivers might even be licensed from UltiMaker, so it's unclear how much control Prusa has over their source code. Plus, even if they were open source they need to be signed which is a something that only an employee at Prusa can do.

Hopefully, this PR motivates someone internally at Prusa to take lead on those aspects that the OSS community can't help with :)

lexcyn, 2026-09-01:
Really disappointing to see the new 3.0 slicer with NO arm64 support. This was the perfect time to implement this, guys.

leohumnew, 2026-09-03:
Created a feature request discussion for this as suggested by the Prusa team. Please upvote it if interested as they seem to be looking to gather community sentiment from the discussions there!

https://github.com/prusa3d/PrusaSlicer/discussions/15727

SachCZ, 2026-09-18:
Thank you @leohumnew, using discussions for this kind of issue is the goal. I am closing this issue, since it is not a reproducible bug.

closed 2026-09-18 by SachCZ