prusa3d/PrusaSlicer #12293: Ramping lift causing zero feedrate moves (Invalid speed in 'G1 X56.922 Y125.318 Z.7 F0' on klipper)

gabevenberg, 2024-02-20:
### Description of the bug

If ramping lift travel is turned on in prusaslicer, it will emit g1 moves with a feedrate of 0 at the beginning of each travel move.

### Project file & How to reproduce

[f0.zip](https://github.com/prusa3d/PrusaSlicer/files/14337654/f0.zip) .3mf file and resulting gcode provided. Make sure 'ramping lift' is turned on.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

[f0.zip](https://github.com/prusa3d/PrusaSlicer/files/14337654/f0.zip)

### Operating system

Arch linux, kernel 6.7.4

### Printer model

Modified ender 3, klipper, direct drive, belted z.

LimesKey, 2024-02-20:
Happens to me too with ramping lift, `Invalid speed in 'G1 X94.924 Y79.738 Z.55 F0'

lukasmatena, 2024-02-20:
Closing as a duplicate of #12258. It is caused by setting "z travel speed" to zero, which should use normal speed, but erroneously uses the literal value. It will be fixed in the next release.

closed 2024-02-20 by lukasmatena