prusa3d/PrusaSlicer #12294: Crash on slice EVERY single time

jcleek, 2024-02-20:
### Description of the bug

Crashes on slicing with ANY model when organic supports are selected. Slicing with GRID and SNUG only work when Prusa Slicer is run as administrator. When slicing it gets to the infill or supports portion it just closes. No error message. Formatted my drive, reinstalled Windows 11 with all updates. Installed Prusa Slicer and immediately saw the exact same behavior.

Computer is no slouch:
Processor	13th Gen Intel(R) Core(TM) i9-13900K   3.00 GHz
Installed RAM	32.0 GB (31.7 GB usable)
System type	64-bit operating system, x64-based processor

### Project file & How to reproduce

Well, I can't. 

File size too big: 25 MB are allowed, 114 MB were attempted to upload.

So here is the file on OneDrive: https://getcis-my.sharepoint.com/:u:/p/jcleek/ETm0W-8nK7NBsSz5xpM91c0BaQm3xykU5XO97TWAkvCVLA?e=HG5IEe

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.7.1+win64

### Operating system

Edition	Windows 11 Pro Version	22H2 Installed on	‎2/‎16/‎2024 OS build	22621.3155 Experience	Windows Feature Experience Pack 1000.22684.1000.0

### Printer model

MK4

lukasmatena, 2024-02-20:
Thanks for the report, but we would need a 3MF containing your configuration so we can try to reproduce it. List of STL files is not enough.

However, it is quite possible that you are really suffering the same fate as the people in #11845. You can try to browse through that thread and try come of the things that have helped others.

jcleek, 2024-02-20:
You are 100% correct and I have posted my findings there based on your test recommendations. Thank you!

closed 2024-02-20 by jcleek

ayeohdee, 2024-06-28:
I have this same problem with a high end setup. I disable e-cores via task manager / set affinity for PS and the crashing stops. With the 13900k I only leave 0-7 enabled and I do not crash at 20% or at any percent anymore.