prusa3d/PrusaSlicer #12296: Unable to start a print

mffguld2022, 2024-02-20:
### Description of the bug

13:23:18 
!! {"code":"key112", "msg": "Move exceeds maximum extrusion (0.878mm^2 vs 0.640mm^2)
See the 'max_extrude_cross_section' config option for details", "values": [0.878, 0.640]}

### Project file & How to reproduce

[20mmTestCube_repaired.zip](https://github.com/prusa3d/PrusaSlicer/files/14344400/20mmTestCube_repaired.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer Version:   2.7.1+win64 

### Operating system

Windows 10

### Printer model

Creality Ender 3 S1 Pro running Klipper

rtyr, 2024-02-20:
It does not look like PrusaSlicer-related issue.

> See the 'max_extrude_cross_section' config option for details", "values": [0.878, 0.640]}

You should configure this parameter in the Klipper properly.

mffguld2022, 2024-02-20:
Thanks for your quick reply 👍I’ll will check my Klipper settingsSent from my IpadBest regardsKent Strobel20 feb.

closed 2026-09-29 by prusaslicer-bot[bot]