prusa3d/PrusaSlicer #12304: PrusaSlicer 2.7.1 - Random Seam Alignment Issue

arushb03, 2024-02-21:
### Description of the bug

When slicing with the random seam position option selected, a few of the perimeters are sliced below the bed plane. If I turn this back to aligned, it works perfectly fine

[image]

### Project file & How to reproduce

[Error.zip](https://github.com/prusa3d/PrusaSlicer/files/14360558/Error.zip)

Just slice with the current config. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1 +win64

### Operating system

Windows 11

### Printer model

Creality Ender 3 (modded)

closed 2024-02-21 by arushb03

SachCZ, 2024-02-21:
It was the the `G1 Z-3 E1.95 F800` was it ? :smile:

arushb03, 2024-02-21:
It was, but I'm not sure why because I had told it `G1 Z3 E-2 F800` just prior to this, so it should just go back to where it was before...