prusa3d/PrusaSlicer #12307: Fixing organic supports to speed print time, reduce material usage and headache

Ert2030, 2024-02-22:
**A clear and concise description of what the problem is....**

I am am EXTREMLEY frustrated when using supports even when painted on! some branches DO NOT NEED TO EXIST! they are printed for absolutely no reason and are not even close to where they need to be, or are not needed at all because the area is already supported by other supports.

**A clear and concise description of what you want to happen and how it would work...**

Add a feature that enables the user to add or delete support branches in locations as needed. Add a feature that enables the user to use the paint on supports feature to PAINT ON THE PRINTER BED, this way the user can select what locations they want supports to come from!!!!

Better yet add a support editor in general. it may be better to do a complete overhaul.  

**A clear and concise description of any alternative solutions or features you've considered...**

Don't have any alternatives... YOU CANT FIGHT AI GENERATED SUPPORTS.... it would be nice to edit them though!

**Additional context**

**Please look at pictures!!!**

[image]
[image]

rtyr, 2024-02-22:
Please provide 3MF project file.

Ert2030, 2024-02-22:
all i have is a stl. sorry. also it wont let me post it. how can i get it to you?

rtyr, 2024-02-22:
You can provide different example if this particular stl is problematic.

Ert2030, 2024-02-22:
> You can provide different example if this particular stl is problematic.

I dont have any other great examples like this one. this site wont let me most any file for 3d printing.

here is a link to the same file I uploaded on thiniverse

https://www.thingiverse.com/thing:6498019

beng27, 2024-02-22:
Have downloaded your STL and sliced. Stuctural 0.2 + organic supports, build plate only. I do see a couple of unecessary branches. A few points: 
- They are very small, and I think are hardly going to affect your filament consumption or print time. 
- I think they are an inevitable byproduct of any algorithm that is searching for overhangs. , Given the small size of the object realtive to nozzle and layer size, a fractional width (and thus non-printing) bit of perimeter is ending up needing to be partially supported. It's possibly a pixelation/resolution problem.
- Slicer says 8 errors auto repaired when your stl is imported. That might be the issue also.
[image]

beng27, 2024-02-22:
[beltLink_supportissue.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14376797/beltLink_supportissue.3mf.zip)
 You have to zip files to upload them. Pretty old-school, I know. I've done that for you.

Ert2030, 2024-02-22:
I do not have any zip files. this is a production run of 10,000 parts possibly more. ANY savings in print time and material consumption is critical for me. this is not the only print I have had these issues with but this one was noteworthy due to the sheer amount of needless branches.

my main question is is it at all possible to fix this issue? knowing prusia is a big company is it possible the feature i am asking about gets added?

closed 2026-09-29 by prusaslicer-bot[bot]