prusa3d/PrusaSlicer #12313: Space mouse settings can't be applied to PS

yoshimitsuspeed, 2024-02-23:
### Description of the bug

I cannot use default space mouse settings that come from the perspective of manipulating the object. I use inverted controls where you are essentially manipulating the camera. 
If I go into 3Dconnexion settings and try to invert settings nothing changes. Even if I disable everything in the settings nothing changes. It is like PS is not looking to 3Dconnextion settings and I see no way to adjust these settings within PS. 

Please make it so I can invert controls in PS or so that 3D connexion settings can be applied to PS. 

### Project file & How to reproduce

Described above

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Win 10

### Printer model

any

yoshimitsuspeed, 2024-04-05:
> This might help: #10701 And possible this: [#7549 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/7549#issuecomment-1006359089) And maybe even more: https://github.com/prusa3d/PrusaSlicer/issues?q=3dconnexion

I don't see any solutions to my issue. 
My mouse does work in the sense that I can navigate around in the PS window. Selecting or deselecting legacy devices doesn't seem to make any difference. 
I am running a legacy device on the latest version of 3D connexion I can 10.6.4. It works in all my other apps and I can invert the controls in everything I have tried. Just not in PS.

TheGreenMamba, 2025-05-12:
Same problem here.

I just want to revert all the axis, because my brain is not used to "grab the object" but more to "rotate the camera around the object".

For this to work, i need all axis inverted.

SuperSlicer has implemented this and it works perfect for me:

[image]

Thanks in advance!

yoshimitsuspeed, 2025-05-14:
> You can still use the 3Dx settings to invert axis:
> 
> [image]

Maybe it is because of 3Dx's horrible planned obsolescence and the fact that I am using an older version of their drivers and software but the last time I tried any changes I made in the 3dX settings had no impact on behavior in PS.

TheGreenMamba, 2025-05-14:
As far as I understand, PS is using the mouse directly without 3dx. 
I'm on linux and changing settings in spacenavd also doesn't change settings in PS.

foreachthing, 2025-05-15:
> because of 3Dx's horrible planned obsolescence

Don't get me started on that...!!!!

Anyway, even with my old hardware and driver (10.8.3.3389) it's working as expected. I can inverse any axis I wish.

[image]

TheGreenMamba, 2026-09-02:
> You can still use the 3Dx settings to invert axis:

On Windows this might be the case, but on Linux it's controlled by PS itself, like it is with SuperSlicer and OrcaSlicer.

The only way to invert the axis on linux is to implement it in PS.