prusa3d/PrusaSlicer #12315: Application Terminates when configuration wizard is run and something is changed

nobowen, 2024-02-24:
### Description of the bug

First, on application start whether profiles from previous versions are loaded or not, all printer presets are available in the drop down menus on the right. application also starts with errors loading configuration files.  When the configuration wizard is run I can only see checkboxes on my printer (MK2s). When I make a change, for example unselecting the 0.25 nozzle, then finish the wizard the application terminates with "Unhandled unknown exception; terminating the application"

This error happens for me on two computers, both my laptop and desktop for any version over 2.6.1 (but not including)
I tried 2.7.0 2.7.1 and a prerelease 2.7.2

### Project file & How to reproduce

[image]
above is the error application starts with

below is the application terminating after trying to change things in the config wizard
[image]

to reproduce, make any change in the configuration wizard and click finish. doesn't happen when no change to the config wizard

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0 and every version higher

### Operating system

Windows 11

### Printer model

Prusa mk2s

Jan-Soustruznik, 2024-03-12:
Hello, @nobowen , could you send us the configuration folder of PrusaSlicer for analyze. Folder is located in `AppData/Roaming/PrusaSlicer`. Just zip the folder "PrusaSlicer" and share in this issue ticket. Thank you

nobowen, 2024-03-16:
[PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/14625503/PrusaSlicer.zip)
Hello, here is the folder you requested

nobowen, 2024-05-30:
I solved the issue, in the appdata/roaming/prusaslicer/printer, my old configs had a value:
thumbnails = 0x0,0x0

which was out of range. To fix I deleted 0x0,0x0 from each printer config

closed 2026-09-29 by prusaslicer-bot[bot]