prusa3d/PrusaSlicer #12316: [BUG] Dependency build fails on Win10

vovodroid, 2024-02-24:
### Description of the bug

I'm struggling with building dependency. I run .`\build_win.bat -d "..\PrusaSlicer-deps" -s deps`, after some time it failed with Boost compilation errors, but it was difficult to find first error . So I opened **PrusaSlicer_deps.sln** in VS2022 and tried to build with "Build solution", it failed with first error

```
>C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.38.33130\include\vcruntime.h(386,51): error C2065: 'uintptr_t': undeclared identifier [C:\DISK_D\3D\Prusa\PrusaSlicer\deps\build\builds\Boost\libs\iostreams\boost_iostreams.vcxproj]
>  (compiling source file '../../../../dep_Boost-prefix/src/dep_Boost/libs/iostreams/src/file_descriptor.cpp')
```

So now I have no possibility to build PrusaSlicer.

### Project file & How to reproduce

Try to build dependency.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2-rc1

### Operating system

Win10x64

### Printer model

Custom

vovodroid, 2024-02-25:
Last version that is built successfully is 2.7.0-alpha1. Commit [df0b3b095bdf99d07dfb94e898f7228b424e9e18](https://github.com/prusa3d/PrusaSlicer/commit/df0b3b095bdf99d07dfb94e898f7228b424e9e18) introduced new folder structure and make file changes, causing failures.

vovodroid, 2024-06-15:
I still can't build deps for 2.8.0, a lot of errors like

```
C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.40.33807\include\vcruntime.h(387,51): error C2
065: 'uintptr_t': undeclared identifier [C:\DISK_D\3D\Prusa\PrusaMaster\deps\build\builds\Boost\libs\iostreams\boost_io
streams.vcxproj] [C:\DISK_D\3D\Prusa\PrusaMaster\deps\build\dep_Boost.vcxproj]
```

Build command:
`.\build_win.bat -d "..\PrusaSlicer-deps-280a3rel" -c Release -p low -s deps`

KimmoHop, 2024-07-18:
For me, dependencies for 2.8.0 went initially fine, but building the slicer ended up in out of heap space.
After updating Visual Studio CE 2019 to latest version (to maybe fix heap problems), building dependencies stops at first stages with
`CUSTOMBUILD : CMake error : Invalid value used with --target [C:\dev\PrusaSlicer\deps\build\deps_debug.vcxproj]`
(there is no value, so CMAKE or something else is not using value from _DESTDIR_)
CMAKE was not updated. CMAKE is oldish version (to avoid #11860). I'm not sure if there is anything to lose...

Last time I built Prusa Slicer was 2 years ago (2.5.0), and build process was smooth (at least compared to Cura) and always working (similar to SuperSlicer build around that time, with instructions that worked with PS too).

Maybe we should start using github actions (e.g. https://github.com/Felix-Rm/PrusaSlicer/actions though slower trial&error process  + no debugging) instead :/

vovodroid, 2024-07-19:
It seems that problem was in msys64 in PATH.

closed 2024-07-19 by vovodroid