prusa3d/PrusaSlicer #12318: no retraction indicator visible when "wipe while retracting" is enabled

cordilon, 2024-02-24:
### Description of the bug

I've been seeing this issue for a couple of PS versions: If the "wipe while retracting" feature is enabled in print settings, there is no indicator for the retraction visible in the sliced preview of the model. The retraction is still carried out and working as intended.
After disabling the feature, the indicator is visible.

[image]
[image]

This is just a basic cylinder with standard Prusa profiles for print, material and printer.
Sorry if this issue was already posted, I couldn't find it via search.

### Project file & How to reproduce

[no_retract_point_visible.zip](https://github.com/prusa3d/PrusaSlicer/files/14392890/no_retract_point_visible.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.2-rc1+win64

### Operating system

Windows 10

### Printer model

Prusa i3 Mk3 (modified)

rickbeec, 2024-03-02:
Same. Version 2.7.2+ win64. Have not verified that retraction is still happening. I was troublehooting another issue where, on one continuous seam, sometimes there is a retraction, and sometimes not.

SachCZ, 2024-03-04:
Hello @cordilon. I do not think there should be an indicator when there is a wipe instead of retraction in place. Do you think otherwise? Thank you for your effort!

cordilon, 2024-03-04:
Hi @SachCZ , I actually do think so, mostly because the wipe moves can be hard to read on more complex layers, so that their direction and end points are unclear. The attached screenshot shows that, and also the fact **some** retractions have indicators, which should be proof that this is a bug and not intentional.

[image]

Another huge benefit of the indicators (as compared to wipe move lines) is that you can see where they are on the outer perimeters when looking at a sliced model, which makes troubleshooting retraction artifacts a lot more straightforward.

By the way, the length of the wipe moves is also all over the place, but I think this has at least been partially fixed in the final release of 2.7.2.

cordilon, 2024-03-05:
@SachCZ , first of all, I did provide a project file (no_retract_point_visible.zip) in the original post.
Also just for you I went all the way back to PrusaSlicer 2.4.2 and sliced the exact same model. Here's a screenshot plus project file just in case, in my opinion this is what it is supposed to look like. As you can see, there are also clipped wipe moves here, but the retraction indicator is still there.
If you only show retractions where the wipe move wasn't completed, the feature seems pretty useless. A wipe move is not the same as a retraction, you could in theory even have wipes without retraction, that's why it never occurred to me that combining them visually could be intentional.

[image]
[PSv2.4.2_retraction_points.zip](https://github.com/prusa3d/PrusaSlicer/files/14500843/PSv2.4.2_retraction_points.zip)

SachCZ, 2024-03-28:
Hello @cordilon, I am sorry I must have missed the original project file, thank you for your patience. I will look into it.

SachCZ, 2024-04-11:
Hello @cordilon, I have went through both 3mfs in both 2.7.2 and 2.4.2. **I can now confirm with confidence that this is not a bug.** Static retractions are displayed as purple blobs, retractions while wiping are displayed as yellow lines. In version 2.4.2 Prusa slic3r was doing some static retractions after every wipe. I removed those using a regexp in prusa slic3r version 2.4.2 and as expected the wipes are still there but the purple dots are not.

The only difference is that 2.7.2 does not generate any static retractions (meaning it retracts fully during the wipe). **This is correct as that is the purpose of the wipe!**

Here is a [static_retractions_removed.zip](https://github.com/prusa3d/PrusaSlicer/files/14951311/static_retractions_removed.zip) generated in 2.4.2 with the mentioned regex to prove my point. Closing.

closed 2024-04-11 by SachCZ

cordilon, 2024-04-13:
Hey @SachCZ , thanks for taking the time and looking further into it. I just checked and you are right: v2.4.2 did tiny static retractions after wipe moves, even if there would have been space to wipe further. Still, I can't think of a reason why, besides making the purple point visible with the forced static retraction... meaning that it wasn't a bug in that version either.
Besides that, it's of course not optimal in terms of printer movement.

My point still stands though: It is difficult to read visually, especially when looking at multiple layers at the same time. I think there should be an option to have both the line to indicate the wipe and the point to indicate the end of the wipe move. As you suggested, I'll probably open a feature request when I find the time =)

SachCZ, 2024-04-15:
Hello @cordilon, thank you for understanding. **A feature request is a good idea!**