prusa3d/PrusaSlicer #12321: Program quit after slicing in case of object duplication

ncandelier, 2024-02-24:
### Description of the bug

When duplicating an object and slicing the program quits systematically : 
Please find below the Windows event : 
[image]
[image]

### Project file & How to reproduce

Please remove .zip extension : 

[Rose_1_Color_Stem.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/14394069/Rose_1_Color_Stem.stl.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version: 2.7.1+win64

### Operating system

Windows 11 (Version 22621)

### Printer model

MK3S+

kubispe1, 2024-02-26:
I am not able to reproduce it. @ncandelier, Do you have the same problem with 2.7.2-rc1?

ncandelier, 2024-03-02:
Sorry for the delay, yes, but this occurs randomly. Might be an issue on my side. I will keep testing and let you know. Thank you.

kubispe1, 2024-03-04:
Okay, I'm closing this issue for now. If you need more information, please let us know, and we'll be happy to reopen it. Additionally, please use the most recent releases whenever possible. Thanks

closed 2024-03-04 by kubispe1