prusa3d/PrusaSlicer #12323: Center camera on center of part

bbolgradov, 2024-02-25:
**Is your feature request related to a problem? Please describe.**
When printing multiple parts it's a bit cumbersome to center the camera on a specific part to look around it when it's not on the center of bed.

**Describe the solution you'd like**
It would be useful to have a shortcut to align camera center with part center.

**Describe how it would work**
For example pressing Space would align camera center with part center.
Space with a modifier (Shift/Option/Alt/Ctrl) would change the behavior to center on the center/center, bottom/center, top/center of part, etc.

**Describe alternatives you've considered**
Currently I switch to top-view and center on XY axis, switch to side-view and center on XZ/YZ. Then the camera center is mostly aligned with the center of the part and you can rotate around it.

buzzhuzz, 2024-03-09:
Try selecting the object and press 'z' key. After that keys 1-6 will switch view port to a side views.

5jvm0u4, 2025-03-01:
OrcaSlicer is defaulted to rotate around object when it's seleted, hope PrusaSlicer could merge this feature back from that branch.

5jvm0u4, 2026-09-03:
Hope this get implemented in PS3.