prusa3d/PrusaSlicer #12324: Toolpath outside of the print area with 210mm x 210mm x 6.75mm with no brim/no skirt

vrosu, 2024-02-25:
### Description of the bug

I have an object 210mm x 210mm x 6.75mm that I'm trying to print on my MK4 (FW: 5.1.3+13503) with PrusaSlicer 2.7.1.
From my experience with my previous MK3S it is not a complicated object - apparently no supports needed etc.
No skirt, no brim, with or without IS profile this results in the same thing.
**I am not able to slice it without obtaining briefly the "a toolpath outside the print area was detected".**

What exactly is wrong with this object? I am able to slice a cube with 210x220x250 (not sure about the order, but the idea is that it's max print volume) and it works without the error above, but with this object I get this...
[image]
[image]

### Project file & How to reproduce

[gf5x5_base_bug.zip](https://github.com/prusa3d/PrusaSlicer/files/14395828/gf5x5_base_bug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 11 Home

### Printer model

MK4 with FW: 5.1.3+13503

mix579, 2024-02-25:
FIle sliced w/o errors using PS 2.7.2-rc1, so I assume they fixed the error (there's mention of something like this in the release notes)

vrosu, 2024-02-25:
thanks @mix579 . I will wait for stable 2.7.2 and retest this. For awareness, even if Prusa Slicer generated this message, the print finished successfully, at least at a first glance at my object.

vrosu, 2024-04-03:
Just to close. After 2.7.2 this issue no longer happens. Safe to close.

closed 2024-04-03 by vrosu