prusa3d/PrusaSlicer #12326: No Wipe Is Present When Outer Walls Printed First

cubitik, 2024-02-25:
### Description of the bug

If print outer walls first is checked, there is  no wipe happening at seams. Works correctly with inner walls first.
[image]
[image]

### Project file & How to reproduce

Slice any model without print outer walls first unchecked, examine seams to see the wipe operation. And slice the same model with just print outer walls first option checked and boom, see, wipe is gone on seams (no wipe).
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 10

### Printer model

ELEGOO Neptune 4 Pro

cubitik, 2024-02-25:
Weirdly, turning on "Minimum travel after retraction" option and setting to 0 in setting overrides of filament profile fixes the issue no wipe at seams with "External perimeters first". This should be fixed.
[image]

Jan-Soustruznik, 2024-02-26:
Hello, @cubitik, 

Could you share your project file - 3MF, thank you

cubitik, 2024-02-26:
> Hello, @cubitik,
> 
> Could you share your project file - 3MF, thank you
[project-files.zip](https://github.com/prusa3d/PrusaSlicer/files/14410390/project-files.zip)