prusa3d/PrusaSlicer #12327: Top and bottom layers wrong pattern if layer count is more than >1

MegaBlackJoe, 2024-02-25:
### Description of the bug

When the number of top or bottom layers is more than 1, the first layer is printed ignoring the settings of the slicer. For example, if you choose 2 top layers and concentric as the pattern for these layers, the frist top layer will have diagonal lines instead. Only the second layer will have the necessary pattern, but not the first. It's also true for bottom layers and the only difference there is that the right pattern will be printed first and then the pattern becomes wrong

### Project file & How to reproduce

Print settings > Infill > Top fill pattern > Concentric

Layers and perimeters > Solid Layers > Top layers = 2

or 

Layers and perimeters > Solid Layers > Bottom layers = 2
[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 11

### Printer model

Elegoo Neptune 4

neophyl, 2024-02-25:
The pattern chosen is only supposed to apply to the very top most or bottom most layer. The rest is normal solid infill.  That’s how it’s designed to work and how it’s always worked.

MegaBlackJoe, 2024-02-25:
the only problem is that you can't choose the pattern of this "normal solid infill" as you call it.

closed 2026-09-29 by prusaslicer-bot[bot]