prusa3d/PrusaSlicer #12329: Multi Material Perimeters

coreyfro, 2024-02-25:
I am a new owner of a Prusa XL.

I would love to be able to have options to make perimeters with multiple materials. Would also include top and bottom surfaces.  I can think of dozens of use cases.

- High cost finish material for the external 2 or 3 perimeters, low cost for any internal.
- Weak external material, stronger internal
- Glow material or other abrasive on outside, minimizing wear on nozzles
- Transparent materials on outside for additional effects (skin, eyeballs, multi-layer color, blending)
- Blended color or other gradients.
- Complimentary high performance materials (tough PC on the outside, rigid PC-CF on the inside: Increases later adhesion with more isotropic PC to compliment the weaker layer adhesion of the PC-CF)
- honeycomb complimentary materials such as alternating PC and PC-CF such that the PC bonds the PC-CF filament together more strongly since the PC-CF will always have PC neighbors.  There can be a ratio of one vs the other.
`0*0`
`*0*`
`0*0`

`0****0`
`******`
`**00**`
`******`
`0****0`

- TPU external and PETG internal for tough or soft overmolding.
- Minimizing jamming from Problematic materials such as woodfill PLA by using it only for the surface.
- Difference sized nozzles for external perimeters for greater detail
- Ability to make ultra thin parts by having breakaway support internal perimeters (print in place nested shells)

tfulkerson, 2026-09-11:
I also would really like to see this implemented. I am wanting to print with a PETG-CF infill and internal perimeters with 1 or 2 external perimeters in normal PETG. This would allow a stronger part while still being safe to handle and not having to worry about the Carbon Fibers becoming micro-splinters.

I have just taken a quick look at the PrusaSlicer 3.0.0 alpha and it does look to be a great update, but I don't currently see support for this feature. There is a feature to have material #1 for the internal fill and material #2 for the perimeter, but that is not really the same thing.

Prusa if you see this it would be amazing if you added this to the Multi-Material section and it could look something like this:

If Perimeters = 2
   Multi-Material Section
      1st Perimeter: <Extruder #1> ( note: 1st material would be the outermost perimeter )
      2nd Perimeter: <Extruder #2>

If Perimeters = 3
   Multi-Material Section
      1st Perimeter: <Extruder #1>
      2nd Perimeter: <Extruder #2>
      3rd Perimeter: <Extruder #2> ( note: this selection would be hidden if Perimeters < 3 )

If Perimeters = 4
   Multi-Material Section
      1st Perimeter: <Extruder #1>
      2nd Perimeter: <Extruder #1>
      3rd Perimeter: <Extruder #2>
      4th Perimeter: <Extruder #2> ( note: this selection would be hidden if Perimeters < 4 )

If Perimeters = n
   Multi-Material Section
      1st Perimeter: <Extruder #1>
      2nd Perimeter: <Extruder #3>
      3rd Perimeter: <Extruder #7>
      4th Perimeter: <Extruder #1>
      ...
      nth Perimeter: <Extruder #2>

This feature could also be an amazing addition to create sandwiched perimeters to create very interesting results. One example might be to have a hard outer shell (PETG), then a softer middle perimeter (TPU), then a hard inner shell (PETG). This could lead to some great impact resistant prints, or lead to other great ideas. See the example below:

If Perimeters = 3
   Multi-Material Section
      1st Perimeter: <Extruder #1> (PETG)
      2nd Perimeter: <Extruder #2> (TPU)
      3rd Perimeter: <Extruder #1> (PETG)

Sembazuru, 2026-09-17:
It seems Orca has added something similar somewhat recently. See this YT short from the NextLayer channel for a valid use-case:
https://www.youtube.com/shorts/PO5HiQy7bcM

To: Prusa mods - You may want to move this to Discussions if you don't make (haven't made) this an internal ticket to comply with the new issue tracker policy...