prusa3d/PrusaSlicer #12330: Z drop over part, not purge tower with No Sparse Layers selected

halofx, 2024-02-25:
### Description of the bug

The print head drops over the part, instead of over the Purge Tower, when using No Sparse Layers setting for Wipe Tower, and traditional Travel Lift. This results in the nozzle drilling into the part, and then trying to drag the part. Issue happens when using multi part prints, and when using an M600 manual change. This does not happen if a Ramping Lift is used. And seems tied to the Lift setting for the next extruder.

In Alpha2 the head doesn't drop, but also doesn't lift off the part, it ramps off the part regardless of setting.
In Beta and RC1, the nozzle lifts as it should, but then drops into parts before moving off the part.

[image]

### Project file & How to reproduce

Attached project is a simple model with 3 stacked boxes, 2mm thick.
Each box is assigned to a different single nozzle extruder.
Extruder 1, and 3 use traditional lift. Extruder 2 uses a Ramping Lift.
A wipe tower is enabled, with No Sparse Layers enabled.
The issue can be seen at the end of the first object, just after ;AFTER_LAYER_CHANGE ;2.2.
A "G1 Z.6" is inserted just before the head moves off the part.
But is doesn't happen at the end of the second object, at the ;4.2 later change.

[Shape-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/14397634/Shape-Box.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.2-beta1+win64

### Operating system

Win10 

### Printer model

Voron 2.4, and Prusa MK4

lukasmatena, 2024-02-26:
Reproduced, thanks. SPE-2176

FidelCapo, 2024-03-01:
This issue should be fixed and the fix is part of the 2.7.2 release.
Closing.

closed 2024-03-01 by FidelCapo