prusa3d/PrusaSlicer #12332: Pruasaslicer 2.7.2-rc1 Crashing Windows 10

colinb88, 2024-02-26:
### Description of the bug

When dragging and dropping a multi-body AMF file into Prusaslicer 2.7.2-rc1, Windows hangs and requires a restart to clear. If the same file is opened via the file, open dialog, a dialog pops up reporting a problems with Windows and the program closes. This just started happening with AMF files created with Solidworks 2016. Never had a problem before.
[PickBox-TeleFender.zip](https://github.com/prusa3d/PrusaSlicer/files/14398345/PickBox-TeleFender.zip)

### Project file & How to reproduce

[PickBox-TeleFender.zip](https://github.com/prusa3d/PrusaSlicer/files/14398363/PickBox-TeleFender.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.2-rc1

### Operating system

Windows 10

### Printer model

Mk4

kubispe1, 2024-02-26:
PS 2.6.0 works to me. We will check it. Thanks

lukasmatena, 2024-02-26:
Reproduced, thanks. SPE-2175

1973darrell, 2024-02-29:
windows 10 hp saids its corrupt software will not let me download 2.7.2

lukasmatena, 2024-02-29:
Fixed in [2.7.2-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-rc2). Closing.

I have just uploaded the 2.7.2 final release. It is possible that you are getting false positive reputation alert by some AV engine. I had updated the binaries to VirusTotal database before the release, but it apparently takes time before it propagates. I am almost sure that it will be fine in a couple of hours/days.

closed 2024-02-29 by lukasmatena