prusa3d/PrusaSlicer #12333: Model Cut - deleting unwanted connectors

GregFax, 2024-02-26:
PrusaSlicer - 2.7.1 
When I cut a model, I usually use spare pieces of filament as the joiner dowels (1.75mm x 10mm long).
But PrusaSlicer automatically creates dowel pins which cannot be deleted from the Plater (without getting an error message).
And I'm sure others might want to use different materials for their dowel pins.

Could we get another option in the Cut menu that says "Don't print dowel pins"?
Alternatively, can you fix deleting them from the plater?

YuSanka, 2024-02-26:
Thanks for your report. But, if you want, you can set any object unprintable.
[image]

closed 2024-02-26 by YuSanka

GregFax, 2024-02-26:
Thanks, that helps. But if you try to Delete an object, you get:
[image]
A bit counter-intuitive...