prusa3d/PrusaSlicer #12335: No mk3.5 non input shaper

cozmo14047, 2024-02-26:
### Description of the bug

Hi, there are a few of is wondering why is there no non inpitshaping profile for mk3.5. Manu thanks

### Project file & How to reproduce

Try to add a preset for mk3.5 there is no non input shaper

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 10

### Printer model

Mk3.5

rtyr, 2024-02-26:
Non-IS MK3.5 never existed (unlike MK4, XL and MINI). MK3.5 FW supports input shaper from the beginning.

closed 2024-02-26 by rtyr

cozmo14047, 2024-02-26:
> Non-IS MK3.5 never existed (unlike MK4, XL and MINI). MK3.5 FW supports input shaper from the beginning.

So we will mever jave a choice to use non input shaping

rtyr, 2024-02-26:
Input shaper is a FW feature, not slicer feature. IS is enabled on the printer side no matter which profile is used. 

Some printers have temporarily two sets of profiles available, since users can use older non-IS firmware.

cozmo14047, 2024-02-26:
> Input shaper is a FW feature, not slicer feature. IS is enabled on the printer side no matter which profile is used.
> 
> Some printers have temporarily two sets of profiles available, since users can use older non-IS firmware.

Ok thanks,

vidgameking, 2024-02-28:
This is unacceptable. All input shaping profiles regardless of printer model limit layer height to only 0.10mm. 

There is no 0.3mm draft mode
There is no 0.07mm Ultra Detail
There is no 0.05mm Ultra Detail

So if temporary non-IS profiles are going away, are IS profiles with 0.05mm layer height going to be added?

I have a MK3.5 that literally is not in use because I cannot print due to new limitations.

mix579, 2024-02-28:
That's important. I ordered a Mk3.5 upgrade assuming all previous configs would be available. I use this particular printer exclusively for one production model that requires a 0.8 mm nozzle and 0.6 mm layer height. If that's not supported, I just wasted a couple hundred bucks. But better to know now than finding out after the upgrade.

rtyr, 2024-02-28:
1) MK3.5 is compatible with MK3S g-codes, so you can use any profile you were using with MK3S.

2) 
> There is no 0.3mm draft mode

0.3mm layer height does not make much sense anymore with IS (unless high flow nozzle is used), because you are hitting maximum volumetric speed already with 0.2mm SPEED profile. 0.30mm layer height would have to use slower speeds than 0.2 to melt filament fast enough.

> There is no 0.07mm

It is possible that we will add lower layer heights than 0.10mm for 0.4mm nozzle if there will be demand for it, but so far it was not the case and it is quite rarely used/needed in general.

3)
> I use this particular printer exclusively for one production model that requires a 0.8 mm nozzle and 0.6 mm layer height. 

You can still print like this and you can even use the same g-code. That being said, there was never stock profile for 0.6mm layer height. The range of supported layer heights is the same with MK3.5.

MK3.5 actually have more profiles in general, since it supports more nozzle diameters.

mix579, 2024-02-28:
I just installed the Mk3.5 profile and I see a lot of layer heights available for the 0.8 nozzle so I'm good. I guess I could also just run my old mk3s gcode files.

mix579, 2024-02-28:
Makes sense, rtyr. When I said 0.6, I guess I was rounding up 😇

rtyr, 2024-02-28:
> I can confirm that both MK4 bgcode and plain MK3S+ gcodes present me with "Not sliced for this Machine" error on my MK3.5.

There is a message telling the user that the gcode was not originally sliced for MK3.5, but the message also tells you it will be printed in MK3-compatibility mode. So the printer knows for which printer it was sliced and also knows, how to modify some parameters. I can understand that this message can be annoying for someone.

> Why is there currently a MK4 non-input shaper profile, since that's effectively native on that printer? I would imagine non-input shaper profiles for printers that have been upgraded from a MK3S+ to sub MK4 (like the MK3.9 or MK3.5) It seems that it is meant to allow co-existence of both Input Shaper and Standard

MK4 printer was initially released without input shaper (it was not ready at the moment), therefore lot of users had (and may still have) a firmware, which does not support input shaper. MK3.9 and MK3.5 never existed in non-IS variants, therefore non-IS profiles were never needed and never existed.

> All non-Input shaper profiles will be removed in a future date, will there be a long-term support version for legacy Prusa Slicer and profiles for users not wishing to use Input Shaper?

Usually only the latest PrusaSlicer versions are getting new profile updates. We may discuss internally adding some new presets.

.

maxkueng, 2024-03-13:
@rtyr If this is really the way this was intended, to use the MK3S+ profile for the MK3.5, then this feels pretty half baked.

The printer profile name should reflect that it works with the MK3.5 (i.e. rename it to Original Prusa i3 MK3S & MK3S+ & MK3.5) and there should definitively not be an annoying warning on the screen.