prusa3d/PrusaSlicer #12339: last position of old layer is the next position in the new layer

SebKoAIM, 2024-02-26:
### Description of the bug

Since the update 2.7 of Prusa Slicer there will allways be a G1 command after the Layer change with the last position of the old layer (just without the E-Value) 
[image]
wich is kind a problem when implementing custom layer change G-Codes

### Project file & How to reproduce

[issue.zip](https://github.com/prusa3d/PrusaSlicer/files/14406440/issue.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 10

### Printer model

ExAM510

SachCZ, 2024-02-26:
Hello @SebKoAIM, I am struggling to reproduce your issue. First of all the 3mf you provided does not seem match the screenshot you provided. Please next time include a screenshot of the 3mf you are attaching.

Now, in your gcode I see no problem with the layer changes, as the move labeled `move to first layer point` (see screenshot) is required for the print to be successful and cannot be omitted.
[image]

Your case might be more special, it seems like after your original layer change (before the substitutions and without any customization) there is no travel in xy required, but I struggle to verify that as the example you provided does not match the screenshot and is cluttered with substitutions and custom gcodes.

Can you please provide more simplified example, ideally with some default settings, tweaked only as much as it is required to reproduce the issue?

I am sorry, but in the current state of the bug report there is not much we can do for you. 

Thank you for your time!

SebKoAIM, 2024-02-27:
Hello and thank you for the fast answer. I'll try to explain the problem more clearly:

**G1 X107.447 Y111.178 E.03902**   ; Last move of the previous layer
_M204 P3000
M486 S-1
M106 S127.5_
;LAYER_CHANGE
_;Z:0.3
;HEIGHT:0.1_
_G1 E-.7 F2100_                ;Retract at Layer Change
_G1 Z.3 F720_                   ;move Z (next Layer)
G92 E0                           ; Custom G-Code to reset E
G1 X5.000 Y5.000 F15000   ; Custom G-Code (just an example move)
_M486 S0_
**G1 X107.447 Y111.178 F18000**    ; same Position as before, in this case an unnecessary move
G1 X134.818 Y112.571                       ; new start point of the next path (in my case the position i want)
G1 E.7 F2100
M204 P2500
;TYPE:Perimeter
;WIDTH:0.449999
G1 F8925.391
G1 X106.983 Y112.571 E.49592

This was not the case with the previous version (2.6.1) and if you open the 3mf with this version it still works as intended.
Thank you for your time!
[issue_2.zip](https://github.com/prusa3d/PrusaSlicer/files/14415606/issue_2.zip)
[image]

SachCZ, 2024-02-27:
Hello, I might have good news for you @SebKoAIM . I was able to reproduce your issue in 2.7.1. Even better, the issue does not seem to present anymore in the latest release candidate, so it is probably already fixed. Could you please download the latest version here https://github.com/prusa3d/PrusaSlicer/releases (at the time of writing the latest version is 2.7.2-rc2). Thank you for your cooperation.

SebKoAIM, 2024-02-28:
Hello,
just tested some different projects and the issue is gone with the latest release. Thank you for the great support

closed 2024-02-28 by Jan-Soustruznik