prusa3d/PrusaSlicer #12340: Incorrect time estimation on 0.2 layer height

NoodleSploder, 2024-02-26:
### Description of the bug

When I select 0.2 layer height for the build plate I have currently I get 6h 54m time estimation.  
When I modify only the layer height to 0.21 for the same build plate I get 1h 25m time estimation.

I am not sure if its doing anything different in the gcode yet, will have to look at it later.

### Project file & How to reproduce

Setup a build plate with 0.2 layer height print setting.  Generate G-Code and look at the time estimate.  Change the print setting for layer height to any other value and see that somethings not right.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.2-rc1+win64-202402231108

### Operating system

Windows 10

### Printer model

Voron Switchwire

NoodleSploder, 2024-02-26:
Files.

[Dummy 13.zip](https://github.com/prusa3d/PrusaSlicer/files/14410843/Dummy.13.zip)
[image]
[image]

rtyr, 2024-02-27:
Autospeed is very old and very experimental feature. We are not planning to extend this feature further.

I suggest to rather use classic perimeter generator instead of arachne, it will likely help in this case.

NoodleSploder, 2024-02-27:
Got it!  I was just experimenting with auto speed - I didnt know it was dead - that is useful to know!  Thanks for following up.

closed 2024-02-27 by NoodleSploder