prusa3d/PrusaSlicer #12341: In the "absorb heat" phase, the nozzle temperature is too high, leads to oozing.

LuckyResistor, 2024-02-27:
### Description of the bug

In the printer profiles for the Prusa XL, during the "absorb heat" phase, the nozzle temperature is set too high for PETG. This leads to unnecessary oozing, which affects the subsequent bed-level calibration.

## Affected Profiles

- Original Prusa XL - 5T 0.4 nozzle
- Original Prusa XL - 5T 0.6 nozzle
- Original Prusa XL - 5T 0.8 nozzle
- Original Prusa XL - 5T Input Shaper 0.4 nozzle
- Original Prusa XL - 5T Input Shaper 0.6 nozzle
- Original Prusa XL - 5T Input Shaper 0.8 nozzle

## Problem Code

The issue is found in the following G-code line:

```
M109 T{initial_tool} S{((filament_notes[initial_tool]=~/.*HT_MBL10.*/) ? (first_layer_temperature[initial_tool] - 10) : (filament_type[initial_tool] == "PC" or filament_type[initial_tool] == "PA") ? (first_layer_temperature[initial_tool] - 25) : (filament_type[initial_tool] == "FLEX") ? 210 : (filament_type[initial_tool]=~/.*PET.*/) ? 175 : 170)} ; wait for temp
```

Here, a *generic* temperature is set, based solely on the filament type. This approach does not account for specific filament variations (like low-temperature PETG) or nozzle sizes.

## Proposed Solutions

### 1. Use the Idle Temperature from the Filament Settings

During the "absorb heat" phase, the nozzle temperature should be set to the idle temperature defined in the filament profile.

### 2. Park the Tool While Waiting for the Temperature

To prevent further oozing from the nozzle, the tool should be parked while waiting for the temperature to stabilize during the "absorb heat" phase.

### Project file & How to reproduce

[project.zip](https://github.com/prusa3d/PrusaSlicer/files/14417669/project.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.2-rc1+MacOS-arm64

### Operating system

macOS 14.3.1 (23D60)

### Printer model

Prusa XL 5T

LuckyResistor, 2024-02-29:
@rtyr Yes, the nozzle needs to be sufficiently heated during the probing process. However, I see no necessity to maintain that temperature for six or more minutes. The nozzle can quickly return to operational temperature within seconds.

Perhaps "oozing" isn't the most accurate term. There's no bubbling, and the process is quite slow. I experience this effect with perfectly dried Prusament PETG Signal White. As detailed in my bug report, the occurrence varies depending on the filament used and the nozzle size.

With the same Prusament PETG Signal White:
- I have no oozing with a 0.4 mm nozzle.
- I have oozing with the 0.6 mm nozzle.
- I have even more oozing with the 0.8 mm nozzle.

The photo below shows how the 0.6 mm nozzle looks after three minutes at 175ºC with Prusament PETG Signal White. This filament that hangs out of the nozzle is then pressed up to the nozzle sides while probing and ends in the print as large, sometimes burned blobs.

[image]

If maintaining a generic 175ºC temperature serves a specific technical purpose, then park the printer head  while waiting in the "absorb heat" phase would be perfectly acceptable.

As a temporary solution, I've manually reduced the temperature to 160ºC during the "absorb heat" phase. This small adjustment works perfectly fine and does not affect the probing process.

I encourage you to test this for yourself:

Please load some freshly dried Prusament PETG Signal White filament into a PrusaXL equipped with either a 0.6 mm or 0.8 mm nozzle. Move the printer head 10 mm over the printer bed. Set the nozzle temperature to 175ºC and the printer bed temperature to 85ºC. I'm betting that you will also witness how filament slowly escapes the nozzle.

rtyr, 2024-02-29:
Thanks for your detailed feedback. We may discuss some possibĺe improvements with FW team. 
As far as I know, the tool must be picked before calling the heat absorbtion phase, because absorbtion time is based on the filament type assigned to that particular tool.

wafflecart, 2024-03-10:
I also get this with PLA on a 0.4mm nozzle getting worse with larger 0.6mm and 0.8mm nozzles. It will **still ooze at 170c**, thus messing up probing and leaving small bits of filament all over the bed and nozzle. The temperatures while waiting to probe needs a bit of tweaking, if it didn't ooze and it was somehow able to keep the nozzle clean before probing it would make things more automated. 

Similar to you I have this manual process I use for PLA:

- load filament like regular
- once filament is loaded, go into settings and reduce temp to 150c this stops the oozing after a few seconds instead of 5 minutes..
- park tool
- then start a print
- when tool comes out of dock it will still have to heat up to 170c before probing bed but oozing will be significantly reduced.

rtyr, 2024-03-21:
The start sequence was updated as follows (only for https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-alpha1 for now, but will be likely also added to stable releases):

1) preheat to "MBL" temp (for example 175C for PETG)
2) home all axis (we don't want to home Z with cold nozzle)
3) set hotend temperature to `idle temperature` without waiting (idle temp is set in filament settings, usually 70C)
4) at the same time, heat absorbtion phase begins, while the nozzle is cooling down
5) after heat absorbtion (or if it is canceled by user), the temp is set back to "MBL" temperature

This is also usable for single tool printers.

closed 2024-03-27 by rtyr