prusa3d/PrusaSlicer #12346: Drag & Drop of multiple 3mf files.

i0n0s, 2024-02-27:
### Description of the bug

When dropping multiple 3mf files from the windows explorer into the PrusaSlicer, only one file will get loaded.
When dropping multiple stl files then all dropped files will get loaded.

### Project file & How to reproduce

[ProjectFile.zip](https://github.com/prusa3d/PrusaSlicer/files/14424264/ProjectFile.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 10

### Printer model

MK4

Jan-Soustruznik, 2024-02-29:
Hello, @i0n0s ,

that is expected behaviour. When you Drag & Drop of multiple 3mf files into open PrusaSlicer, then is open only one file. (3MF file = project file; .stl = model, geometry). Exception is if you try to Drag & Drop .zip file with multiple 3MF files selected, then geometry is loaded only. 

For more info see the article on our block page about the 3MF file: https://blog.prusa3d.com/3mf-file-format-and-why-its-great_30986/

closed 2024-02-29 by Jan-Soustruznik

i0n0s, 2024-02-29:
I understand that it doesn't make sense to load multiple profiles from 3mf files, but there are also 3mf files just containing a model.
And I can already add the (model) content of multiple 3emf files into Prusa Slicer by dropping them one after another. As a user, I just expect dropping multiple files to behave the same as multiple drops of files.

Jan-Soustruznik, 2024-03-01:
@i0n0s In this case you can use "Add..." (CTRL+I) and import models only from multiple 3MF file without load any settings, if any.