prusa3d/PrusaSlicer #12347: Print stops and asks to change filament multiple times during printing.

ciaocibai, 2024-02-27:
### Description of the bug

When slicing with 2.7.1 the printer will randomly stop and ask me to change filament. There is no M600 in the gcode. Downgraded to 2.6.1 and the issue is no longer present.

### Project file & How to reproduce

github won't let me upload the gcode for the print. But this is happening consistently with all prints I've done since updating to 2.7.1

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 10

### Printer model

Prusa i3 Mk3

rtyr, 2024-02-27:
Your issues are most likely related to filament sensor. You can easily confirm that by disabling the filament sensor on the printer side.

We can check the gcode if you provide it (you have to archive it to the zip first), but I don't think there will be any issue with the gcode.

ciaocibai, 2024-02-27:
Thanks for the response. I have tried disabling the sensor - didn’t resolve the issue. However rolling back to version 2.6.1 did solve the issue. I see from a support thread on the forums the someone else is having the same issue as well. Will upload gcode when I’m back at my computer.Sent from my iPhone

ciaocibai, 2024-02-28:
Here is the last gcode that produced this issue for me. Note I've had the same issue with multiple prints on 2.7.1, but it doesn't appear on the same file sliced with 2.6.1

[Plug thicknesser (1)_0.3mm_PLA_MK3_43m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/14431555/Plug.thicknesser.1._0.3mm_PLA_MK3_43m.gcode.zip)

rtyr, 2024-02-28:
Your g-code is ok, I also printed it on MK3 without issues. I still think it's about filament sensor (especially if you have older "non-S" MK3).

Please contact our tech support via live chat at www.prusa3d.com, they will help you with troubleshooting.

I am closing the issue here, the problem is not related to PrusaSlicer.

closed 2024-02-28 by rtyr