prusa3d/PrusaSlicer #12348: When starting 2.7.2 RC2 - PrusaSIicer has encountered an error

stevekenney318, 2024-02-27:
### Description of the bug

When starting 2.7.2 RC2 - PrusaSIicer has encountered an error

[image]
[image]
[image]

### Project file & How to reproduce

not sure what to put here ....

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.2-rc2+win64

### Operating system

Microsoft Windows 11 Home 10.0.22631 Build 22631

### Printer model

Mingda Magician X2

lukasmatena, 2024-02-28:
Could you please zip your config folder (Help->Show configuration folder) and post it here? Thanks.

stevekenney318, 2024-02-28:
> Could you please zip your config folder (Help->Show configuration folder) and post it here? Thanks.

Hi Lukáš, 

I seemed to have gotten past the issue by creating a new configuration snapshot. But I went back to the previous one with the error and 'zipped' the config folder - it is the one named 'PrusaSlicer.12348' . I'm also including the one that works , and it is called 'PrusaSlicer.good' Since it is working fine for me now, it is no longer a critical situation. I would still be interested to know what caused it though. 

Thanks, Steve

[PrusaSlicer #12348.zip](https://github.com/prusa3d/PrusaSlicer/files/14432904/PrusaSlicer.12348.zip)
[PrusaSlicer good.zip](https://github.com/prusa3d/PrusaSlicer/files/14432907/PrusaSlicer.good.zip)

lukasmatena, 2024-02-28:
The config folder looks that it was massively hampered with. As if someone copied one folder over another and chose to keep files from both, just renaming duplicates. It is beyond my imagination how PrusaSlicer itself could achieve something like this. Were you doing some manual transfer of older profiles before the rc2 was started?

lukasmatena, 2024-02-29:
Thanks for the reply. Since I am not a native English speaker, I struggled to find a better wording to exactly describe my impression.
I am closing the issue with the assumption that it was caused by the innocent tweaking going wild.

closed 2024-02-29 by lukasmatena

stevekenney318, 2024-02-29:
I think you do quite well for English not being your native language !!  I thank you for your time and effort on this matter !

stevekenney318, 2024-02-29:
@lukasmatena 

Hi Lukáš, I just wanted to let you know that 2.7.2 installed without issue !! Thanks.