prusa3d/PrusaSlicer #12351: PrusaSlicer 2.7.2-rc2 UI looks broken

pmjdebruijn, 2024-02-28:
### Description of the bug

PrusaSlicer 2.7.2-rc2's UI looks badly broken, even though it's still technically entirely functional:

[image]

This is locally built on Debian 12 "Bookworm" with wxWidgets 3.2.2, with greybird-gtk-theme, this is also the default theme on Xubuntu, so it's in fairly common use.

### Project file & How to reproduce

Set the greybird GTK theme
Build PrusaSlicer 2.7.2
Start PrusaSlicer

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.2-rc2

### Operating system

Debian 12

### Printer model

Ender 3

pmjdebruijn, 2024-02-28:
For additional clarity:

I'd expect the background of active fields, like ComboBoxes/SpinBoxes/EditFields to be white, as a grey background often signal a disabled widget, so this ends up being quite confusing. Additionally this lowers contrast in the UI which isn't beneficial either.

I'd expect the white edges around the checkboxes to not be there, this is purely cosmetic.

lukasmatena, 2024-02-28:
Does it happen with our AppImages downloaded from Releases page here on GitHub?

pmjdebruijn, 2024-02-28:
No clue, I built the sources myself in the form of a Debian package, because I really don't want to run precompiled binaries.

pmjdebruijn, 2024-02-28:
Though locally built 2.6.1 didn't have this issue yet.

lukasmatena, 2024-02-29:
> I really don't want to run precompiled binaries

That is very nice, but if your locally-built PrusaSlicer is the only one where the issue manifests, there is nothing we can do.

pmjdebruijn, 2024-02-29:
I just tried the following AppImage in a xubuntu 23.10 live cd in a VM:
https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.7.2-rc2/PrusaSlicer-2.7.2-rc2+linux-x64-GTK3-202402261525.AppImage

As expected the issue is exactly the same.

pmjdebruijn, 2024-02-29:
If you can point me to the relevant lines of code, I'm willing to experiment a bit on my end...

YuSanka, 2024-03-01:
@pmjdebruijn, could you please provide a screen shot of the Plater, please?

closed 2026-09-29 by prusaslicer-bot[bot]