prusa3d/PrusaSlicer #12355: Setting Maximum Fan Speed

martinj999, 2024-02-29:
When printing models with flat surfaces, e.g.  a box , when printing the final surface layers the fan is set to maximum (255) this is hard coded in the gcode file. There is a Min and Max in the settings but this is not used in the gcode file? With the fan at full speed I get a THERMAL RUNOFF ... How to set the Max Speed of the fan that is used in the gcode file? Maybe a bug in the Min / Max settings not picking up on set value?

martinj999, 2024-02-29:
Hi

I don’t have 3mf files… I’ve attached the stl  of the model in question. When I create gcode the fan speed is set to 255 by the slicer for certain layers, my settings in slicer have max speed set to 10% .

 

Am I doing something wrong? 

 

 

Best regards

 

Martin

 

Please provide your 3MF project file.

rtyr, 2024-02-29:
We need your 3mf project file (can be saved from prusaslicer) or at least the gcode so we can see your setrings.

rtyr, 2024-02-29:
Also you cannot attach files via email, you need to zip the file first and then upload it via github web interface. We don't need stl.

martinj999, 2024-02-29:
[CounterWeight-BodyCylinder002.zip](https://github.com/prusa3d/PrusaSlicer/files/14453131/CounterWeight-BodyCylinder002.zip)

rtyr, 2024-02-29:
I see. The first solid layer printed on sparse infill is printed as a bridge. Bridges have its own fan speed setting, which is not limited by "max fan speed" setting. 

Bridge fan speed is set to 100% in your settings, you can reduce it if you want.

That being said, if you have Original MK3 printer and you are getting thermal runaway errors during PLA prints, than you have likely some HW issue on your printer.

martinj999, 2024-02-29:
The only time it happens is when the fan goes full speed as descried, otherwise it's working fine. If I change the bridge speed I presume all will be OK?

martinj999, 2024-02-29:
I modifed the gcode file manually and it printed out OK...

rtyr, 2024-02-29:
What I am saying is that standard MK3 printer should handle 100% fan speed without trigerring thermal runaway error.

Anyway, I suppose we can close this issue?

martinj999, 2024-02-29:
thanks for your help, I've only been using it for a few weeks, getting used to it's quirks lol :)

closed 2024-02-29 by rtyr