prusa3d/PrusaSlicer #12357: Wrong PrusaSlicer-2.7.2+MacOS-universal-202402291330 release file?

ulab, 2024-02-29:
### Description of the bug

The MacOS release file contains a folder called "Original Prusa Drivers" which then has additional links to New user guides and Sample G-Code pages.

[image]

This seems wrong and while I do like that you now ship with a directly linked `Applications` folder, I'd rather only have the `PrusaSlicer.app` directly in my `Applications` folder and not "Drivers" which are none?

### Project file & How to reproduce

Screenshot is linked above

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

MacOS

### Printer model

XL5

ulab, 2024-09-19:
It's this way again in 2.8.1

ahti, 2024-09-24:
Yeah this seems like a Windows-ism brought over to the Mac distribution to me. The DMG should just have the PrusaSlicer.app and a link to /Applications, no folders containing links or any such stuff.

closed 2026-09-29 by prusaslicer-bot[bot]