prusa3d/PrusaSlicer #12358: "Fasten" option to allow object to be broken into two parts with optional fastener types to reconnect them

jgoldwein, 2024-02-29:
**Is your feature request related to a problem? Please describe.**
At times, I wish to print an object that is larger than the print plate of my printer or that would otherwise need to be split into two objects that could be re-fastened together.  For example, a top and bottom of a pill box for a smaller object or a long cylinder or rectangle that exceeds the dimensions of the printer.

**Describe the solution you'd like**
For said object, a tool to split the object into two separate objects that fasten back together using a variety of possible fastening types would solve the problem.  In this way, each of the two objects created by the tool could be printed separately and then re-fastened together to create the larger object.  In addition, the tool could be used to create fasteners for smaller objects (e.g. pill box) in a way that would be easier than designing the fasteners using CAD programs such as Fusion 360..

**Describe how it would work**
Once the entire object has been imported into the slicer, a tool would allow the user to select the point at which the object should be split.  Next, the user would select the fastener type to be used from a set of fastener type options.
 
**Describe alternatives you've considered**
The option to this is to design the fastener type in the CAD program used to design the object.

**Additional context**
Add any other context or paint drawing about the feature request here.

jmd73, 2024-02-29:
@jgoldwein I think you are talking about the cut tool that already is in PrusaSlicer since a long time already.  Check the documentation [here](https://help.prusa3d.com/article/cut-tool_1779)

jgoldwein, 2024-02-29:
I am embarrassed that I did not realize connectors were a part of the cut tool.   Very sorry about that.  

I do wonder if other connectors such as keystone/dovetail type?  I am guessing there are others as well.

closed 2024-03-01 by SachCZ