prusa3d/PrusaSlicer #12359: Automatic Arrange creates overlapping

3DPrintMunich, 2024-02-29:
### Description of the bug

When using support and auto arraging parts there were still overlaps caused, which result in an error message
[image]
[240222 AICI Plateau 1 - Wheel - Outer.zip](https://github.com/prusa3d/PrusaSlicer/files/14452654/240222.AICI.Plateau.1.-.Wheel.-.Outer.zip)

### Project file & How to reproduce

1) Auto Arrange
2) Slice Parts
3) Get error message

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

Win10

### Printer model

Mk3s

neophyl, 2024-03-01:
Please search for relevant existing issues before opening new ones.
For example this one 
https://github.com/prusa3d/PrusaSlicer/issues/12174
https://github.com/prusa3d/PrusaSlicer/issues/12022
https://github.com/prusa3d/PrusaSlicer/issues/11421
https://github.com/prusa3d/PrusaSlicer/issues/10486
https://github.com/prusa3d/PrusaSlicer/issues/11133
https://github.com/prusa3d/PrusaSlicer/issues/9891
https://github.com/prusa3d/PrusaSlicer/issues/9723

all the way back to
https://github.com/prusa3d/PrusaSlicer/issues/3042
https://github.com/prusa3d/PrusaSlicer/issues/1275
There are many many more too.
The earliest is https://github.com/prusa3d/PrusaSlicer/issues/316 and this issue is where they are all collated.

Some of those other issues detail why the software works the way it does, and has done so for a very long time.

The arrange function gives you the ability to set the distance that objects are spaced.  If you are using supports with first layer expansions like you are, or if using organic supports which tend to produce larger footprints outside of the models then you need to increase the distance to a suitable amount.  The warning was added relatively recently so you the user are aware of the overlap and can do something about it.  

For your objects the simplest would be to change support type to Snug as grid expands outwards more than necessary.   Unless there is a specific reason you are using Grid ?

kubispe1, 2024-03-01:
Hi, I am closing this as a duplicate.
@neophyl, please could you contact me at my email: petr.kubis@prusa3d.cz.
Thanks

closed 2024-03-01 by kubispe1