prusa3d/PrusaSlicer #12361: Wipe tower not starting on first/zero layer

MattBlissett, 2024-02-29:
### Description of the bug

With the new 2.7.2 release, the wipe tower for a multimaterial print is not starting on the first layer.  Filament change layers on the wipe tower are also misaligned vertically.

The preview flashes correctly momentarily, before appearing as in the screenshot.

Note the wide wipe tower base is printed *before* the rest of the tower.

[image]

### Project file & How to reproduce

- Multi-extruder print
- Add cube
- Add cylinder modifier to cube, move to overlap
- Set cylinder to a different extruder

[issue.zip](https://github.com/prusa3d/PrusaSlicer/files/14454085/metric-square-sticks-issue.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.2+linux-x64-GTK2-202402291326.AppImage

### Operating system

Ubuntu 22.04

### Printer model

Mk3S+ Revo

MattBlissett, 2024-02-29:
Screenshot from 2.7.2-rc2:

[image]

I see the only change in the release notes, and the problem is caused by a line in my tool-change G-code:

```
G1 Z{layer_z+2} F720; go up a bit more
```

I had this to prevent coloured dots on the print from the nozzle returning to the previous location.

So probably a very minor issue.

lukasmatena, 2024-03-01:
I did not realize that it is potentially a breaking change for projects like yours.
When your custom G-code changes position or retraction, you should inform PrusaSlicer about it, so the G-code generator is not thrown off track. You can do it by adding `{position[2] = position[2] + 2.0;}` in your custom toolchange G-code.

MattBlissett, 2024-03-01:
That works fine.

With https://github.com/prusa3d/PrusaSlicer/issues/2672 I shouldn't need my custom G-code anyway.

The issue is solved as far as I'm concerned, thanks.

commit fa23ab82 (2024-03-25): Make sure that after wipe tower toolchange the G1 Z is always emitted.

Even if the G1 Z would be redundant, it is still emitted.
Fixes: #12361
closed 2024-03-01 by rtyr