prusa3d/PrusaSlicer #12362: 2.7.2 upload to PrusaConnect get 400 error

rogervanhoutan1961, 2024-03-01:
### Description of the bug

after installing 2.7.2 i can no longer directly upload from prusa slicer fails with the following error
[image]
reinstalling 2.7.1 i get the ability to upload again.
Reboot did not help.

### Project file & How to reproduce

slice the file and click on upload to prusa connect, upload, upload to queue, and upload and print all fail the same way
the file transfers but when it gets to 100% it fails with the attached 400 error.
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

version 2.7.2 +win64

### Operating system

windows 10

### Printer model

mk3s+

MartinMajewski, 2024-03-10:
It gets worse. After I tried to reregister my MK4 on PrusaConnect, I got a 401 error stating that my key was either missing or invalid. However, it is the correct key.

[image]

Interestingly, it works with the Prusa Mini.

rogervanhoutan1961, 2024-03-10:
I uninstalled everything and reset up my prusa slicer and printer settings from scratch it is working now

MartinMajewski, 2024-03-10:
However, that's not a feasible solution if the setup is complex.

MartinMajewski, 2024-04-11:
Push!

There must be a bug in PrusaSlicer.

PrusaLink: Works, but only with "HTTP digest" credentials
Octoprint: Works with API key (which is sent via the HTTP header, I guess?)
PrusaConnect: Nope!

However, the printer is reachable, controllable, and monitorable via PrusaConnect.

PrusaSlicer version 2.7.4

MartinMajewski, 2024-04-11:
I had to remove all my printer presets and redo them in order to get PrusaConnect working! 

It works again, but there is a big bug in PrusaSlicer somewhere. I would be glad to assist finding it.

closed 2026-09-29 by prusaslicer-bot[bot]