prusa3d/PrusaSlicer #12366: 2.7.2+: Post Processing Scripts Broken! (Error Code 1)

CCS86, 2024-03-01:
### Description of the bug

I have been using the same PP script for ages and now I get an error code 1 when trying to export code in 2.7.2+. If I open the same project file in 2.7.0b1 it works just fine.

### Project file & How to reproduce

...

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.2+

### Operating system

Windows 10 x64

### Printer model

UMO+

SachCZ, 2024-03-02:
Hello @CCS86, your description is too general and you did not provide anything that we can work with. Would you please consider uploading a 3mf file? Thank you in advance.

CCS86, 2024-03-02:
It's a really general problem, as I believe it applies to any PP script, not the contents of the script. You guys could certainly try to reproduce by evoking any PP script in 2.7.2 and see if you can reproduce the error. 

Here's a project file.

[Shape-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/14469771/Shape-Box.zip)

SachCZ, 2024-03-04:
Hi @CCS86, the problem is with your specific post processing script. Slic3r 2.7.2 does not emit a layer change at the begging of a print. That broke your post processing script here: `if args.first_layer_start in line:`, as the tag is not present anymore in the gcode. To fix it, move the tag to the start gcode. I did it for you here: 
[Shape-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/14481419/Shape-Box.zip)