prusa3d/PrusaSlicer #12368: configuration issue

gwak64, 2024-03-02:
### Description of the bug

When I load a 3mf file created with a different model than my MK3.5, I lose 3 out of 4 physical printers. Only the MK3.5 with the 0.4 nozzle is then entered in the configuration assistant. The printers with 0.25, 0.6, and 0.8 nozzles must then be re-entered. And sometimes the physical printers also have to be re-entered.

### Project file & How to reproduce

Any 3mf file that was not generated with an MK3.5.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.71 + 2.72

### Operating system

Win 11

### Printer model

Prusa Mk3.5

gwak64, 2024-03-02:
New situation:
Loading from printables.com directly into the PrusaSlicer causes all nozzles except 0.4 to disappear from the configuration. This means that the configuration wizard must be started to set up and the physical printers (except nozzle 0.4) must be set up again.

Jan-Soustruznik, 2024-04-11:
Hello, @gwak64, could you please provide the link to model on printable page where you encountered the issue? Thank you!

gwak64, 2024-04-11:
I can't remember, it was all the models I tried.
But the problem has been solved with version 2.74.
Greetings
Gerd

closed 2024-04-12 by Jan-Soustruznik