prusa3d/PrusaSlicer #12370: Wrong height setting in GCODE  after slice in version 2.7.2

yaronze, 2024-03-02:
### Description of the bug

In Prusa Slicer version 2.7.0 was a bug that I reported – Wrong lift height value in Gcode ([#11843](https://github.com/prusa3d/PrusaSlicer/issues/11843)). 
This bug was supposed to be fixed in version 2.7.2
I downloaded version 2.7.2 and after slicing I am getting this Gcode –

;BEFORE_LAYER_CHANGE
G92 E0.0
;0.4
M201 X3999.84 Y3999.84
M74 W0.151028

G1 E-.7 F2100
G1 Z.4 F720 				**This value is now set to Z.4. Is it possible to modify it?** 
;AFTER_LAYER_CHANGE
;0.4
M104 S220 ; set temperature
M486 S0
G1 Z.6					                       **Height set to Z.6 by the slicer**
G1 X136.868 Y116.868 F18000	                **Goto start printing position**
G1 Z.4 F720				                         **Height set to Z.4 by the slicer**
G1 E.7 F2100
;TYPE:Perimeter
;WIDTH:0.449999
G1 F9479.902
G1 X113.132 Y116.868 E.80343
G1 X113.132 Y93.132 E.80343

It seems to have a new problem now.
1.	Is it possible to modify the value from 0.4 via a parameter in the slicer?
2.	Why after height set to 0.4 it goes up to 0.6 and then goes down to 0.4?
3.	It seems to me that the print will start at 0.4 height instead of 0.2 as it should.
Am i missing something? Or is it a bug?

### Project file & How to reproduce

[Shape-Box_0.4n_0.2mm_PLA_MK4IS_20m.zip](https://github.com/prusa3d/PrusaSlicer/files/14469371/Shape-Box_0.4n_0.2mm_PLA_MK4IS_20m.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Prusa Slicer 2.7.2 official

### Operating system

Windows 10

### Printer model

Prusa MK4

SachCZ, 2024-03-02:
Hi @yaronze, from the provided gcode I can see that you are **refering to lines 2752 - 2755** that correspond to a layer change between layer 1 and 2. Layer 1 is printed with nozzle at z 0.2, layer 2 to is printed at z 0.4 so this gcode seems absolutelly fine to me. Also I recommand you to inspect the gcode in Prusa GCodeviewer. Make sure to toggle the travels in the gcodeviewer and you will se that the first travel is indeed elevated and https://github.com/prusa3d/PrusaSlicer/issues/11843 is resolved. If you think my conclusions are incorrect, feel free to let us know 👍

SachCZ, 2024-03-02:
Just to elaborate a bit on why there is the 0.6 value in the code. It is 0.4 (print z) + 0.2 (retract_lift). The **lift** of the nozzle is performed before any travel. **It can be set** in `Printer settings -> Extruder 1` or in `Filament Settings -> Filament overrides`. You **cannot change the value 0.4** otherwise than setting the **layer_height** parameter in `Print settings -> Layers and Perimeters`.

yaronze, 2024-03-02:
Hi @SachCZ 
I am referring to these lines in GCODE. Please notice that this lines come immediately after MBL and line purge before first layer , so the height should be Z0.2 and NOT 0.4 as in the code.
In earlier versions the height value in line 2553 was taken from retract lift.
The output now from version 272 is -

G1 E-.7 F2100
G1 Z.4 F720
;AFTER_LAYER_CHANGE
;0.4
M104 S220 ; set temperature
M486 S0
G1 Z.6
G1 X136.868 Y116.868 F18000
**G1 Z.4 F720**                                              **Height 0.4 for first layer**
G1 E.7 F2100
;TYPE:Perimeter
;WIDTH:0.449999
G1 F9479.902
G1 X113.132 Y116.868 E.80343
G1 X113.132 Y93.132 E.80343
G1 X136.868 Y93.132 E.80343
G1 X136.868 Y105 E.40172

i am attaching file again - new project
[Box v272.zip](https://github.com/prusa3d/PrusaSlicer/files/14469822/Box.v272.zip)

SachCZ, 2024-03-02:
I have **again** taken your gcode **Box_v272** went through it and than confirmed my suspicion in Prusa GCodeviewer. You are still referening to the layer change between layer 1 and 2. Here is a screenshot of how it looks on my screen:
[image]

The first layer clearly starts at Z0.2. There is some missunderstanding going on. Please send a screenshot of your gcodeviewer where the first layer is supposedly floating.

yaronze, 2024-03-02:
OK. I can see now.  I missed something all the time. By mistake i looked at the 2nd layer.
I just wanted to figure out if the 2.7.0 bug was fixed and missed it again and again.
It seems to be ok.
Sorry for interrupting. 
Thanks for your help.

closed 2024-03-02 by yaronze