prusa3d/PrusaSlicer #12375: Z height is not emitted for wipe tower

EliasG-272168, 2024-03-02:
### Description of the bug

After going from 2.7.1 to 2.7.2, it seems the Z height is not emitted and this results in a wipe tower printed midair.

### Project file & How to reproduce

[image]
<img width="1437" alt="Bildschirmfoto 2024-03-02 um 19 00 21" src="https://github.com/prusa3d/PrusaSlicer/assets/59373347/0c78773a-a570-495e-a4c9-7aa02a

[ProjectFile+Config.zip](https://github.com/prusa3d/PrusaSlicer/files/14470370/ProjectFile%2BConfig.zip)
3f1c67">

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

MacOS 14.3.1

### Printer model

E3D toolchanger RRF

SachCZ, 2024-03-04:
Hello, this is sadly an issue we **unknowingly introduced** but are **already aware** of. It has been **already discussed** here: https://github.com/prusa3d/PrusaSlicer/issues/12361.

The relevant information is that you should **put** `{position[2] = position[2] + 2.0;}` **in your layer change gcode**. (2.0 in this case is the lift performed by your printer firmware, feel free to include any other value).

We did not realize that relying on the user provided position would break projects such as yours. In the hindsight it seems obvious. In the future we will probably return to the previous functionality, but for now I recommend you to inform slic3r about the tool position after the toolchange.

Thank you for your time!

SachCZ, 2024-03-26:
It should be fixed in the latest beta, closing.

closed 2024-03-26 by SachCZ